arendst/Tasmota #2192: Turns on and off by it self

mkomarac, 2018-03-19:
Make sure these boxes are checked before submitting your issue - Thank you!

- Sonoff 4CH PRO
- Firmware 5.12.0

some channels are turning on and off without user interaction. 
I have several Sonoff devices flashed with Tasmota firmware. Sonoff Basic flashed with Tasmota works couple of months without single failure. After this very good experience I have bought two Sonoff 4ch PRO devices and strange things started to happened. Switches are turned on and off by them selfs, without any user interaction. It happens on both PRO's flashed with firmware 5.9.1. I have upgraded firmware to 5.12.0 and problems keeps going. 

Can someone help with this issue?
Does anyone else have similar problem?

I have posted same issue in Home Assistant forum.
https://community.home-assistant.io/t/tasmota-problem/47488
But I believe this is the place to post specific issue.

Thank You all in advance!
BR

fpapale, 2018-03-19:
I have the same problem with a Sonoff Dual. Another same device works right; the firmware is the same: 5.12 for both devices

fpapale, 2018-03-19:
I have the same problem with a Sonoff Dual. Another device works right; the firmware is the same: 5.12 for both devices

mkomarac, 2018-03-19:
it's so irritating lights turning on late at night while watching tv, or going to sleep. 
And every time, light turns ON, manually turning off. 100 times. it drives me crazy. 
I don't know what to do, and I really love Tasmota, but it's not working reliably. I don't want to throw it away, but if the issue is unsolvable I need to revert to old school switches, and I don't want to do that.

vMeph, 2018-03-19:
Have same problem somethimes

arendst, 2018-03-19:
Mine don't

mkomarac, 2018-03-19:
the PianSom on HA forum suggested this solution... 
https://www.youtube.com/watch?v=aq8_os6g13s
I'll try this and report back. 
but issue is also with 2nd device, on workbench without wires soldered...
I'll try to clear 433 MHz memory from PRO's to see will this help...
but somehow I don't believe this suggestions will solve the issue.

wasi54, 2018-03-19:
The Problem is most likely the WiFi
Check if there is no metall around your sonoffs. If the sonoffs are closer than 2" to a metall plate, they will lose the wifi connection, You may try out with sonoffs already working and then put a metall plate beside the sonoff. Also make sure your wify signal is ok  and you do not have other wifis from your neighbor on the same channel. Set the wifi mode to retry instead to restart!
If you have wires soldered, additionally give a chance to the solution from mkomarac/PianSom on youtube.
I have more then 10 sonoffs (without wires soldered) running with 5.12.0 without any issues. Most of the sonoffs are up and running for more than 15 day's.

mkomarac, 2018-03-19:
PRO's are mounted on metal rail!
and cabinet door are also metal!
I was suspicious about wifi signal... this is from Sonoff build into cabinet: AP1 SSId (RSSI) | WiFi (80%)

another PRO is on wooden workbench without metal plates around, and no wires soldered, and having similar issues... 

WiFi mode retry instead restart, is option on Sonoff or Router?

wasi54, 2018-03-19:
Retry is a sonoff mode, see https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
(WifiConfig) or modify user_config.h before compile to
#define WIFI_CONFIG_TOOL       WIFI_RETRY

wasi54, 2018-03-19:
Try to leave the cabinet door open for some time to test !?

wasi54, 2018-03-19:
And a wifi signal of 80% is not a guaranty. Reflection can also be an issue on strong signals.

mkomarac, 2018-03-19:
> Retry is a sonoff mode, see https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
(WifiConfig) or modify user_config.h before compile to
#define WIFI_CONFIG_TOOL WIFI_RETRY

Will You be so kind, and write complete command, so I can paste it into command line in web interface of Sonoff. I haven't compile it on my own with Arduino... I have flashed it with mirko/SONOTA method, so I haven't soldered jumper pins for FTDI adapter.

thanks

mkomarac, 2018-03-19:
[image]
take a look of number of turning on (and I manually each time turn off) today from 06:00 till now

wasi54, 2018-03-19:
http://yousonoff/cm?cmnd=WifiConfig%204

mkomarac, 2018-03-19:
thanks.
after apply this, the issue remains.

wasi54, 2018-03-19:
Did you checked the console output? Debug?

mkomarac, 2018-03-19:
this is from console:

13:35:38 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:35:38 MQT: stat/sonoff4_1/POWER4 = OFF
13:36:11 MQT: stat/sonoff4_1/RESULT = {"POWER1":"ON"}
13:36:11 MQT: stat/sonoff4_1/POWER1 = ON
13:36:14 MQT: stat/sonoff4_1/RESULT = {"POWER1":"OFF"}
13:36:14 MQT: stat/sonoff4_1/POWER1 = OFF
13:36:32 MQT: stat/sonoff4_1/RESULT = {"POWER1":"ON"}
13:36:32 MQT: stat/sonoff4_1/POWER1 = ON
13:36:34 MQT: stat/sonoff4_1/RESULT = {"POWER1":"OFF"}
13:36:34 MQT: stat/sonoff4_1/POWER1 = OFF
13:36:54 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T13:36:54","Uptime":"0T00:40:16","Vcc":3.150,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"WiFi","RSSI":82,"APMac":"F0:B4:29:D8:CC:6C"}}
13:37:27 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:37:27 MQT: stat/sonoff4_1/POWER4 = ON
13:37:53 MQT: Attempting connection...
13:37:53 MQT: Connected
13:37:53 MQT: tele/sonoff4_1/LWT = Online (retained)
13:37:53 MQT: cmnd/sonoff4_1/POWER = 
13:39:56 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:39:57 MQT: stat/sonoff4_1/POWER4 = OFF
13:41:04 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:41:04 MQT: stat/sonoff4_1/POWER4 = ON
13:41:11 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:41:11 MQT: stat/sonoff4_1/POWER4 = OFF
13:41:47 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:41:47 MQT: stat/sonoff4_1/POWER4 = ON
13:42:05 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T13:42:04","Uptime":"0T00:45:16","Vcc":3.151,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"WiFi","RSSI":80,"APMac":"F0:B4:29:D8:CC:6C"}}
13:42:34 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:42:34 MQT: stat/sonoff4_1/POWER4 = OFF
13:42:35 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:42:35 MQT: stat/sonoff4_1/POWER4 = ON
13:42:36 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:42:36 MQT: stat/sonoff4_1/POWER4 = OFF
13:43:01 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:43:02 MQT: stat/sonoff4_1/POWER4 = ON
13:43:08 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:43:09 MQT: stat/sonoff4_1/POWER4 = OFF
13:45:24 MQT: stat/sonoff4_1/RESULT = {"POWER1":"ON"}
13:45:24 MQT: stat/sonoff4_1/POWER1 = ON
13:45:26 MQT: stat/sonoff4_1/RESULT = {"POWER2":"ON"}
13:45:26 MQT: stat/sonoff4_1/POWER2 = ON
13:45:27 MQT: stat/sonoff4_1/RESULT = {"POWER1":"OFF"}
13:45:27 MQT: stat/sonoff4_1/POWER1 = OFF
13:45:28 MQT: stat/sonoff4_1/RESULT = {"POWER2":"OFF"}
13:45:28 MQT: stat/sonoff4_1/POWER2 = OFF
13:45:43 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:45:43 MQT: stat/sonoff4_1/POWER4 = ON
13:45:46 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:45:46 MQT: stat/sonoff4_1/POWER4 = OFF
13:46:00 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:46:00 MQT: stat/sonoff4_1/POWER4 = ON
13:46:01 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:46:01 MQT: stat/sonoff4_1/POWER4 = OFF
13:46:45 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:46:45 MQT: stat/sonoff4_1/POWER4 = ON
13:47:10 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T13:47:10","Uptime":"0T00:50:16","Vcc":3.149,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"WiFi","RSSI":80,"APMac":"F0:B4:29:D8:CC:6C"}}
13:47:57 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:47:57 MQT: stat/sonoff4_1/POWER4 = OFF
13:49:45 MQT: Attempting connection...
13:49:45 MQT: Connected
13:49:45 MQT: tele/sonoff4_1/LWT = Online (retained)
13:49:45 MQT: cmnd/sonoff4_1/POWER = 
13:51:24 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:51:24 MQT: stat/sonoff4_1/POWER4 = ON
13:52:19 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T13:52:19","Uptime":"0T00:55:16","Vcc":3.149,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"WiFi","RSSI":80,"APMac":"F0:B4:29:D8:CC:6C"}}
13:53:30 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
13:53:30 MQT: stat/sonoff4_1/POWER4 = OFF
13:55:32 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
13:55:32 MQT: stat/sonoff4_1/POWER4 = ON

mkomarac, 2018-03-19:
yeah
it's so funny sitting in the living room, watching tv, and light is coming on by itself, and I keep pressing off button.

it reminds me on The Big Bang Theory when some Chinese guys across the World have turned lights @ Lenards home 😄 
same thing is happening to me, my wife is so angry... daughter says it wasn't me... 😄 
Wife Approval Factor is not going to pass if this keeps happening 😃

wasi54, 2018-03-19:
Console output with more debug?

mkomarac, 2018-03-19:
15:06:54 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:06:54 MQT: stat/sonoff4_1/POWER4 = OFF
15:07:22 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:07:22 MQT: stat/sonoff4_1/POWER4 = ON
15:07:31 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:07:31 MQT: stat/sonoff4_1/POWER4 = OFF
15:08:55 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:08:55 MQT: stat/sonoff4_1/POWER4 = ON
15:09:17 MQT: Attempting connection...
15:09:17 MQT: Connected
15:09:17 MQT: tele/sonoff4_1/LWT = Online (retained)
15:09:17 MQT: cmnd/sonoff4_1/POWER = 
15:09:37 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T15:09:37","Uptime":"0T02:10:16","Vcc":3.148,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"WiFi","RSSI":82,"APMac":"F0:B4:29:D8:CC:6C"}}
15:09:59 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:09:59 MQT: stat/sonoff4_1/POWER4 = OFF
15:10:28 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:10:28 MQT: stat/sonoff4_1/POWER4 = ON
15:11:09 MQT: Attempting connection...
15:11:09 MQT: Connected
15:11:09 MQT: tele/sonoff4_1/LWT = Online (retained)
15:11:09 MQT: cmnd/sonoff4_1/POWER = 
15:11:13 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:11:13 MQT: stat/sonoff4_1/POWER4 = OFF
15:13:03 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:13:03 MQT: stat/sonoff4_1/POWER4 = ON
15:13:24 MQT: Attempting connection...
15:13:24 MQT: Connected
15:13:24 MQT: tele/sonoff4_1/LWT = Online (retained)
15:13:24 MQT: cmnd/sonoff4_1/POWER = 
15:13:46 MQT: Attempting connection...
15:13:46 MQT: Connected
15:13:46 MQT: tele/sonoff4_1/LWT = Online (retained)
15:13:46 MQT: cmnd/sonoff4_1/POWER = 
15:14:55 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T15:14:55","Uptime":"0T02:15:16","Vcc":3.149,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"WiFi","RSSI":80,"APMac":"F0:B4:29:D8:CC:6C"}}
15:15:32 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:15:32 MQT: stat/sonoff4_1/POWER4 = OFF
15:16:40 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:16:40 MQT: stat/sonoff4_1/POWER4 = ON
15:17:28 RSL: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:17:33 RSL: stat/sonoff4_1/POWER4 = OFF
15:17:38 MQT: Attempting connection...
15:17:38 MQT: Connected
15:17:38 MQT: tele/sonoff4_1/LWT = Online (retained)
15:17:38 MQT: cmnd/sonoff4_1/POWER = 
15:18:13 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:18:13 MQT: stat/sonoff4_1/POWER4 = ON
15:18:44 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:18:44 MQT: stat/sonoff4_1/POWER4 = OFF
15:20:13 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T15:20:13","Uptime":"0T02:20:16","Vcc":3.150,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"WiFi","RSSI":76,"APMac":"F0:B4:29:D8:CC:6C"}}
15:20:17 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:20:17 MQT: stat/sonoff4_1/POWER4 = ON
15:21:45 MQT: Attempting connection...
15:21:45 MQT: Connected
15:21:45 MQT: tele/sonoff4_1/LWT = Online (retained)
15:21:46 MQT: cmnd/sonoff4_1/POWER = 
15:22:19 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:22:19 MQT: stat/sonoff4_1/POWER4 = OFF
15:23:23 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:23:23 MQT: stat/sonoff4_1/POWER4 = ON
15:25:21 MQT: tele/sonoff4_1/STATE = {"Time":"2018-03-19T15:25:21","Uptime":"0T02:25:16","Vcc":3.148,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"WiFi","RSSI":80,"APMac":"F0:B4:29:D8:CC:6C"}}
15:26:23 MQT: Attempting connection...
15:26:23 MQT: Connected
15:26:23 MQT: tele/sonoff4_1/LWT = Online (retained)
15:26:23 MQT: cmnd/sonoff4_1/POWER = 
15:26:38 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:26:38 MQT: stat/sonoff4_1/POWER4 = OFF
15:27:00 MQT: stat/sonoff4_1/RESULT = {"POWER4":"ON"}
15:27:00 MQT: stat/sonoff4_1/POWER4 = ON
15:27:10 MQT: stat/sonoff4_1/RESULT = {"POWER4":"OFF"}
15:27:10 MQT: stat/sonoff4_1/POWER4 = OFF

arendst, 2018-03-19:
As it's power4 most of the times I suggest, as I did before in other issues, you to clear the RF signals on your Sonoff Pro. They have nothing to do with Tasmota as it's an independent circuit controlled by the microcontroller receiving RF signals and controlling the relays.

Search for clearing RF signals on iTead website and then return here if it still persists.

arendst, 2018-03-19:
As another item it could also be a retained power4 toggle message as you use Hass and I hate Hass as it uses retained messages people just don't understand the consequences of it.

mkomarac, 2018-03-20:
sure, I'll clear RF signals, and report back.
As I want to eliminate 2nd sonoff PRO from interfering, I pressed Reset Configuration in Tasmota menu, and can't access tasmota anymore over network, and not even can't connect to it's own Hot Spot to configure network. it seems it's bricked. And it worked yesterday...
I have flashed Tasmota to original ITEAD software with [mirko/SONOTA](https://github.com/mirko/SonOTA) method. And it worked great. How to flash Tasmota again to unresponsive Sonoff device?

Jason2866, 2018-03-20:
Have you tried the long press button reset method?
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

mkomarac, 2018-03-20:
@arendst 
I have cleared RF codes by long press button to clear RF codes, but it haven't solved the problem...
it did, partially, but now the problem has reflected to switch 3 :(
now, after cleared RF codes, (test time 30 min, yeah, I know, it is too short time to tell anything), but now switch 4 remains off, but switch 3... I select switch 3 to ON mode, and after 8 mins it switches to OFF without any user interaction, just by it self. 
So, one more thing remains to try, is RC filter...
I'll try that as soon as I came home during daytime... and have time to dismantle complete installation...
and of course get back here...
so please, don't close this thread. I hope someone will have use of it!

You were referring to HA and retaining messages.
Would You be so kind, and give more detailed explanation what You mean about it, hating HA, and retaining messages?

I can say, that Home Assistant is exactly what I have looking for whole my life. Affordable, simple and nice way to have smart home. 

thanks

mkomarac, 2018-03-22:
after couple of days testing, it seems that problem is related with MQTT broker. 
any ideas?

Jason2866, 2018-03-22:
See #2140

mkomarac, 2018-04-03:
After lot's of testing, the problem is with manual buttons.

if I push button on the device, (or other push button dislocated away from PRO by STP cable) it won’t retain state, and turn ON/OFF; to be exactly different state, than it was.
but, if I control PRO’s by “soft” switches, HA Interface, HADashboard, or Alexa. everything is working as it should. So maybe problem lies by controlling PRO (or Tasmota) by hard wire switches!?
push button is connected to GPIOx in PRO’s and it triggers relay. I don’t see difference activating relays by pushing hard button, or any other soft button.

Jason2866, 2018-04-04:
PullUp resistors installed?

mkomarac, 2018-04-05:
NO.

myfingerhurt, 2018-05-09:
You may try to add 1K Pull-Up resistors to your GPIOx, might be helpful.

schumi2004, 2018-06-15:
Same issue here with a Basic version after 1 day of perfect working.

Log says this:
2018-06-15 19:12:40.364 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":1,"svalue":"","Battery":55,"RSSI":5}
2018-06-15 19:12:40.482 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":0,"svalue":"","Battery":55,"RSSI":4}
2018-06-15 19:13:07.788 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":1,"svalue":"","Battery":55,"RSSI":5}
2018-06-15 19:13:07.911 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":0,"svalue":"","Battery":55,"RSSI":5}
2018-06-15 19:13:19.407 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":1,"svalue":"","Battery":55,"RSSI":5}
2018-06-15 19:13:19.531 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":0,"svalue":"","Battery":55,"RSSI":5}
2018-06-15 19:14:20.119 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":1,"svalue":"","Battery":55,"RSSI":5}
2018-06-15 19:14:20.239 MQTT: Topic: domoticz/in, Message: {"idx":23,"nvalue":0,"svalue":"","Battery":55,"RSSI":5}

Going to try the Pull-up resistor thing.
During my investigation i disabled all communication with Domitcz in anyway and it still happend, after reboot of the Sonoff Basic ot only got worse.

HobbytronicsPK, 2018-07-12:
I'm having the same issue with Sonoff Slampher running Tasmota 6.1.0. 
I am not even using MQTT and disabled it under settings.
Slampher turns light ON at 5:20 AM in the morning automatically without any user interaction. It also turns light ON several times a day.

ciprian288, 2018-07-12:
Same issue  with Sonoff 4CH pro Tasmota 6.1.0. When i set 2 releys on inching mode and restart my sonoff the inching releys  are turning on . I am not using MQTT .

jasonehines, 2018-07-27:
I have this issue with Sonoff Basic, unmodified (except for Tasmota), with version 6.1.1. I've even tried resetting and reflashing it via the GPIO.

I also have it on Basics that have with switches attached to GPIO 14. They would occasionally turn on by themselves on version 5.14 but it's much worse wit 6.1.1.

HobbytronicsPK, 2018-07-28:
I have experienced ghost switching if I forget to erase flash memory before
flashing Tasmota firmware in my Sonoff devices. Erasing flash memory and
re-flashing the Tasmota resolved the issue in my case.

So always erase the Flash memory before flashing new Sonoff Tasmota
Firmware. Have a look on to the following guide for more information.

https://hobbytronics.pk/sonoff-original-firmware-backup-restore/#Erase-Flash-Memory

Jason2866, 2018-07-31:
Do you use in your broker "retain"? If yes thats your issue.
If not, do you have connected switches via wires? 
If yes disconnect for testing the wires, and see if ghost switchimg is gone.

jasonehines, 2018-07-31:
@Jason2866 I have explicitly set retain to False. The Sonoff is stock, except for the firmware.

HobbytronicsPK, 2018-08-01:
> @HobbytronicsPK <https://github.com/HobbytronicsPK> I erased the flash
> and redid my Sonoff Basic about an hour ago on mine and it's already turned
> itself back on.

Jason, have you turned the MQTT OFF in Tasmota?

Have a look on to
https://hobbytronics.pk/yeti-tasmota-control-sonoff-without-internet/ for
controlling Sonoff without MQTT and Raspberry Pi Server using yeti smart
phone app.

HobbytronicsPK, 2018-08-01:
> No i use only mqtt, that is the great benefit. I choose Tasmota because of
> mqtt.
> mqtt is small and lightweigt communication protocol. I will never use a
> other way...
> Do you use a mqtt broker? How is your complete setup at the moment?
> Something is switching your device

It is a complete setup discussed on the link
https://hobbytronics.pk/yeti-tasmota-control-sonoff-without-internet/.
Please have a look on to it for more information. For a brief overview, it
uses Sonoff Tasmota and Yeti, a smart phone app to control Sonoff devices
directly.

Regards,

*Waqas Ahmed*

*Hobbytronics PK*

Mochiwali, Kahror Pakka

59340, Lodhran, Punjab PK

*Call:* +(92 - 03) 007 3210 41

*Web:* www.hobbytronics.pk

mkomarac, 2018-08-01:
After a while without ghost switching, here we go again!
The Issue is in Sonoff 4ch PRO 
it is powered by 12V DC and switch 3 and 4 are wired with extension STP cables to additional switches
this is switch 3
[image]
and this is switch 4
[image]

Do You think that pull up resistors will permanently resolve problem?

can someone draw schematics for Sonoff 4CH pro where exactly put resistors?
[image]

it would be nice to have complete guide, to help others with same issue

Jason2866, 2018-08-01:
Yes, pullups will (probably) solve your problems. Sorry i dont have a 4ch pro, so cant guide you...

mkomarac, 2018-08-07:
at which device have You had problems? 
Sonoff 4CH Pro?

I am asking because it has no sense at all, that channels 1, 2 and 4 working properly without issue or two in 6 months, and channel 3 is keep switching on and off constantly. It was the time that all has worked properly, over two months, (since HomeAssistant 056 until version 071) after upgrading to 073, 074 issues returned. I have reverted back to 071m but issue remained.
so, right now I have light show in my house, or some ghost switching, which reminds me on one episode of The Big Bang Theory about remote switching lights in students home in Beijing. So they said, that someone switching lights from Pasadena, CA. So funny. But I don't think that Leonard or Sheldon are switching my lights on and off.  I don't know what to do more. pull up resistors are probably solution, but why isn't reflecting to another switch in the same box... so stupid.

ascillato, 2018-08-29:
Hi,

Have you managed to solve your issue?

rPraml, 2018-08-29:
I have a similar issue,
always when the device crashes or reconnect to the mqtt broker (mosquitto 1.4.12) or even if I just restart mosquitto, the sonoff 1ch toggles.

It is 100% reproduceable, on every `sudo /etc/init.d/mosquitto restart` or device reboot, the switch toggles. I als tried to change the PowerRetain and LWT (SetOption10) value, but without success.
Any Ideas what can I try?

```
22:51:45 MQT: Connect failed to 10.75.14.1:1883, rc -2. Retry in 10 sec
22:51:57 MQT: Attempting connection...
22:51:57 MQT: Connected
22:51:57 MQT: tele/sw-schlafzimmer/LWT = Online (retained)
22:51:57 MQT: cmnd/sw-schlafzimmer/POWER = 
22:51:57 MQT: stat/sw-schlafzimmer/RESULT = {"POWER1":"OFF"}
22:51:57 MQT: stat/sw-schlafzimmer/POWER1 = OFF
```

mosquitto.log
```
1535579517: New connection from 10.75.14.82 on port 1883.
1535579517: New client connected from 10.75.14.82 as DVES_8E23BA (c1, k15, u'DVES_USER').
1535579517: Sending CONNACK to DVES_8E23BA (0, 0)
1535579517: Received PUBLISH from DVES_8E23BA (d0, q0, r1, m0, 'tele/sw-schlafzimmer/LWT', ... (6 bytes))
1535579517: Received PUBLISH from DVES_8E23BA (d0, q0, r0, m0, 'cmnd/sw-schlafzimmer/POWER', ... (0 bytes))
1535579517: Received SUBSCRIBE from DVES_8E23BA
1535579517:     cmnd/sw-schlafzimmer/# (QoS 0)
1535579517: DVES_8E23BA 0 cmnd/sw-schlafzimmer/#
1535579517: Sending SUBACK to DVES_8E23BA
1535579517: Sending PUBLISH to DVES_8E23BA (d0, q0, r1, m0, 'cmnd/sw-schlafzimmer/POWER', ... (6 bytes))
1535579517: Received SUBSCRIBE from DVES_8E23BA
1535579517:     cmnd/sonoffs/# (QoS 0)
1535579517: DVES_8E23BA 0 cmnd/sonoffs/#
1535579517: Sending SUBACK to DVES_8E23BA
1535579517: Received PUBLISH from DVES_8E23BA (d0, q0, r0, m0, 'stat/sw-schlafzimmer/RESULT', ... (16 bytes))
1535579517: Received PUBLISH from DVES_8E23BA (d0, q0, r0, m0, 'stat/sw-schlafzimmer/POWER1', ... (3 bytes))
1535579517: Received SUBSCRIBE from DVES_8E23BA
1535579517:     cmnd/DVES_8E23BA/# (QoS 0)
1535579517: DVES_8E23BA 0 cmnd/DVES_8E23BA/#
1535579517: Sending SUBACK to DVES_8E23BA
```

ascillato, 2018-08-29:
@rPraml your case is not ghost switching. 

your log: `22:51:57 MQT: cmnd/sw-schlafzimmer/POWER =` means that Tasmota is receiving that command from your mqtt broker

You are using retain values in your broker.

Please, search in issues for examples on how to solve that.

ascillato, 2018-08-29:
@mkomarac

Hi,

Have you managed to solve your issue?

au190, 2018-08-30:
Hi have you resolved the problem ?

ascillato2, 2018-09-01:
Hi,

Closing issue as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-09-01 by ascillato2

mkomarac, 2018-09-03:
no
the problem is still on

mkomarac, 2018-09-03:
After implementing R-C filter, everything worked great for 3 days.
after that time, problem reappears. but now on the different switch
[image]
so the problem still exists.

more help please.

my lights are working like turn signal on car :(

HobbytronicsPK, 2018-09-03:
I suspect, Tasmota uses button interrupts.

Whenever you will do the stuff via interrupt, you will face this issue even
after adding up RC-Filter.

If a heavy load in vicinity will be toggled, interrupt will detect a false
trigger.

Polling the button state in loop may resolve the issue.

mkomarac, 2018-09-03:
6 LED lamps are per switch 3 and 4 (gu 5.3 led 12v , 2W)
so I don't believe there is too much load on relays

can You please explain, "Polling the button state in loop may resolve the issue."

tnx

MightyMilster, 2018-09-10:
Hi, I also had ghost-switching. Tried adding a pull-up resistor. Replaced old sonoff with a new sonoff. Set Tasmota to "PowerRetain - On". None of this helped and the blinking continued.

I eventually solved my problem and thought I'd add it here in case it helps someone else going crazy like I was. Remember that we are dealing with controlling a high voltage device. When you add a manual switch to GPIO14, you are dealing with a low voltage. This low voltage is usually being carried on very thin wire. What happens when high voltage cables are placed near low voltage cables - Induction!!!! Mysterious unpredictable ghost switching. I suggest keeping the two as far apart as possible and definitely use shielded cable for your switch wires. Good luck!

mkomarac, 2018-09-12:
I am using cat6 shielded cable for manual switching. if that is the case, why another switch works normal without ghost switching?

MightyMilster, 2018-09-12:
This sort of interference can be VERY random and hard to pin-point. cat6 is twisted pairs and the 4 pairs are twisted too. Just takes one point in the cable to be sitting along side a high voltage cable for it to interfere. Can you run a test a test cable to the problem sonoff, which is nowhere near your high voltage cabling? If there are any pressure points where one is crushed/clamped against the other you could try placing folded aluminium foil between high and low voltage cables. Does the cat 6 go all the way through to the PCB? Look for any sections that are not shielded that could be suing from induction.

mkomarac, 2018-09-17:
will do that, and report back

mkomarac, 2018-09-27:
OK
after rearranging wires, I can  say that situation is getting better. but not quite. 
I have also configure tasmota according this video: https://www.youtube.com/watch?v=31IyfM1gygo
and indeed, the situation is getting better. but still, it's not reliable. 
the light still came on randomly, without any interaction.

for example, the most annoying situation, is at late night, just before going to sleep, while watching tv.
the eyes are wide open, adjusted to watching tv, and lights in living room came on. so shocking, annoying. 
I am about to give up, and throw this device away. 
Can't believe, that device is meant to be installed into power cabinet, rail mounted, and according to issues I am experiencing, issues are generated because is mounted into power cabinet.

I'll post picture, to show complete installation. 

is there any more thing I can try. any suggestions?

au190, 2018-09-27:
Can you see át that time disconnect connect of that device in the mosquitto log ?

au190, 2018-09-27:
Log
1537991642: Client BedRoom already connected, closing old connection.

1537991642: New client connected from 192.168.2.16 as BedRoom (c1, k15, u'xcc0').
1537992150: Saving in-memory database to /data/mosquitto.db.

I have mosquitto broker. If you go logs you shoud see something like this. Because the ghost switching can come if your Esp reconnects to the broker and retension is different on the Esp and hassio.

[57 more replies]