arendst/Tasmota #2194: Enhancement: Status5 result with DHCP/static IP info

curzon01, 2018-03-19:
To manage my Tasmota pool I'm using the StatusX results but currently using `Status5` I'm unable to differentiate if IPAddress is a result of DHCP or static setup.

Only using commmand `IPAddress1` immediately returns the desired info but with common `...RESULT` topic, example:
- DHCP IP: `{"IPAddress1":"0.0.0.0 (192.168.17.93)"}`
- Static IP: `{"IPAddress1":"192.168.17.93 (192.168.17.93)"}`

This is hard to eval because all other commands are also using the result topic.

**Request**
Having additonal object in StatusNET json object containing static IP flag, e. g.
- {"StatusNET":{"Hostname":"devtest","IPAddress":"192.168.17.93",**"StaticIP","false"**....
or 
- {"StatusNET":{"Hostname":"devtest","IPAddress":"192.168.17.93",**"DHCP","true"**....

I suggest leaving `IPAddress` format as is because it's better machine readable than "x.x.x.x (x.x.x.x)"  form cmnd `IPAddress1` RESULT.

arendst, 2018-03-19:
Try ``SetOption4 1``

curzon01, 2018-03-19:
Used SetOption4 already as workaround. But this needs to publish and eval an extra topic (currently simply 
use Status 0).
I will see... (may be will try to add it by myself). Thanks a lot, Theo

closed 2018-03-19 by curzon01