arendst/Tasmota #2195: Why are there cmnd retained messages and why are they inconsistent?

wsw70, 2018-03-19:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using: Sonoff Basic x 2 and Sonoff Touch
- [x] Provide the output of command ``status 0``: (see below in context)

I am trying to understand some persistant messages set by my Sonoff devices. I have a Mosquitto broker and when connecting to it I immediately get some messages (retained ones):

    tele/maison/chambre1/lampe_bureau/LWT Online
    tele/maison/chambre1/mur1/LWT Online
    tele/maison/chambre1/lampe_haute/LWT Online
    cmnd/maison/chambre1/lampe_bureau/POWER OFF
    cmnd/maison/chambre1/mur1/POWER ON
    cmnd/maison/chambre1/lampe_haute/POWER ON

The first three ones are understandable - they provide to the newly connected client a status of the devices. 

The next ones are not clear for me: 

- the current state of all the switches is `OFF` 
- why is `cmd` a retained topic? 
- if there is a reason for it to be retained: this is not the last command. Since all of the three switches work in unison (and none was accessed directly), the message is inconsistent (it should be `cmnd/maison/chambre1/<device>/POWER OFF` for all `<device>`).

`status 0` for the device `lampe_haute` is below. Note that it is powered off, the MQTT status above was `ON`:

```
{
  "Status": {
    "Module": 1,
    "FriendlyName": "Sonoff",
    "Topic": "maison/chambre1/lampe_haute",
    "ButtonTopic": "0",
    "Power": 0,
    "PowerOnState": 3,
    "LedState": 1,
    "SaveData": 1,
    "SaveState": 1,
    "ButtonRetain": 0,
    "PowerRetain": 0
  },
  "StatusPRM": {
    "Baudrate": 115200,
    "GroupTopic": "sonoffs",
    "OtaUrl": "http://domus1:80/api/arduino/sonoff.ino.bin",
    "Uptime": "1T01:55:17",
    "Sleep": 0,
    "BootCount": 6,
    "SaveCount": 113,
    "SaveAddress": "FB000"
  },
  "StatusFWR": {
    "Version": "5.12.0",
    "BuildDateTime": "2018-02-09T14:54:51",
    "Boot": 6,
    "Core": "2_3_0",
    "SDK": "1.5.3(aec24ac9)"
  },
  "StatusLOG": {
    "SerialLog": 2,
    "WebLog": 2,
    "SysLog": 0,
    "LogHost": "domus1",
    "LogPort": 514,
    "SSId1": "WSRV",
    "SSId2": "indebuurt2",
    "TelePeriod": 300,
    "SetOption": "00000009"
  },
  "StatusMEM": {
    "ProgramSize": 485,
    "Free": 516,
    "Heap": 23,
    "ProgramFlashSize": 1024,
    "FlashSize": 1024,
    "FlashMode": 3
  },
  "StatusNET": {
    "Hostname": "maison/chambre1/lampe_haute-339",
    "IPAddress": "10.100.10.114",
    "Gateway": "10.100.10.254",
    "Subnetmask": "255.255.255.0",
    "DNSServer": "8.8.4.4",
    "Mac": "68:C6:3A:A3:2D:44",
    "Webserver": 2,
    "WifiConfig": 2
  },
  "StatusMQT": {
    "MqttHost": "10.200.1.1",
    "MqttPort": 1883,
    "MqttClientMask": "DVES_%06X",
    "MqttClient": "DVES_A32D44",
    "MqttUser": "DVES_USER",
    "MAX_PACKET_SIZE": 1000,
    "KEEPALIVE": 15
  },
  "StatusTIM": {
    "UTC": "Mon Mar 19 11:31:15 2018",
    "Local": "Mon Mar 19 12:31:15 2018",
    "StartDST": "Sun Mar 25 02:00:00 2018",
    "EndDST": "Sun Oct 28 03:00:00 2018",
    "Timezone": 1
  },
  "StatusSNS": {
    "Time": "2018-03-19T12:31:15"
  },
  "StatusSTS": {
    "Time": "2018-03-19T12:31:15",
    "Uptime": "1T01:55:17",
    "Vcc": 3.122,
    "POWER": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "WSRV",
      "RSSI": 78,
      "APMac": "4E:D9:E7:6B:74:29"
    }
  }
}
```

The same information for device `lampe_bureau`, which MQTT status is OFF (and is off as well below):

```
{
  "Status": {
    "Module": 1,
    "FriendlyName": "Sonoff",
    "Topic": "maison/chambre1/lampe_bureau",
    "ButtonTopic": "0",
    "Power": 0,
    "PowerOnState": 3,
    "LedState": 1,
    "SaveData": 1,
    "SaveState": 1,
    "ButtonRetain": 0,
    "PowerRetain": 0
  },
  "StatusPRM": {
    "Baudrate": 115200,
    "GroupTopic": "sonoffs",
    "OtaUrl": "http://domus1:80/api/arduino/sonoff.ino.bin",
    "Uptime": "0T23:02:13",
    "Sleep": 0,
    "BootCount": 6,
    "SaveCount": 80,
    "SaveAddress": "F4000"
  },
  "StatusFWR": {
    "Version": "5.12.0",
    "BuildDateTime": "2018-02-09T14:54:51",
    "Boot": 6,
    "Core": "2_3_0",
    "SDK": "1.5.3(aec24ac9)"
  },
  "StatusLOG": {
    "SerialLog": 2,
    "WebLog": 2,
    "SysLog": 0,
    "LogHost": "log.swtk.info",
    "LogPort": 514,
    "SSId1": "WSRV",
    "SSId2": "indebuurt2",
    "TelePeriod": 300,
    "SetOption": "00000009"
  },
  "StatusMEM": {
    "ProgramSize": 485,
    "Free": 516,
    "Heap": 23,
    "ProgramFlashSize": 1024,
    "FlashSize": 1024,
    "FlashMode": 3
  },
  "StatusNET": {
    "Hostname": "maison/chambre1/lampe_bureau-42",
    "IPAddress": "10.100.10.116",
    "Gateway": "10.100.10.254",
    "Subnetmask": "255.255.255.0",
    "DNSServer": "8.8.4.4",
    "Mac": "68:C6:3A:A3:10:A6",
    "Webserver": 2,
    "WifiConfig": 2
  },
  "StatusMQT": {
    "MqttHost": "10.200.1.1",
    "MqttPort": 1883,
    "MqttClientMask": "DVES_%06X",
    "MqttClient": "DVES_A310A6",
    "MqttUser": "DVES_USER",
    "MAX_PACKET_SIZE": 1000,
    "KEEPALIVE": 15
  },
  "StatusTIM": {
    "UTC": "Mon Mar 19 11:41:43 2018",
    "Local": "Mon Mar 19 12:41:43 2018",
    "StartDST": "Sun Mar 25 02:00:00 2018",
    "EndDST": "Sun Oct 28 03:00:00 2018",
    "Timezone": 1
  },
  "StatusSNS": {
    "Time": "2018-03-19T12:41:43"
  },
  "StatusSTS": {
    "Time": "2018-03-19T12:41:43",
    "Uptime": "0T23:02:13",
    "Vcc": 3.174,
    "POWER": "OFF",
    "Wifi": {
      "AP": 1,
      "SSId": "WSRV",
      "RSSI": 62,
      "APMac": "4E:D9:E7:6B:74:29"
    }
  }
}
```

curzon01, 2018-03-19:
Tasmota never publish messages on `cmnd` topics, it only subscribes to it.

Results are published by Tasmota using `stat` topics, not `tele`. `tele` topics are telemetry information. The status of power can be found on `stat` topics

If you have retained messages for `cmnd` topic, it was published by another client.

see also Wiki [MQTT Topic definition](https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features#mqtt-topic-definition)

wongnam, 2018-03-19:
This issue was happen due to some msg are still retained in MQTT broker, in case of Sonoff switchs start to subscribe will lead to MQTT broker will send a retained msg(ON or OFF) to switch immediately.

My case study and workaround for this issue is that I have to clear the retained msg from MQTT broker by following trick: After send a MQTT command ON or OFF to cmnd/sonoff/POWER then get the status of Sonoff switch from stat/sonoff/POWER and change it(the msg) to something else eg: "PowerOnState" and then send it to cmnd/sonoff/POWER again retain this msg in MQTT broker.

I can explain by node-red flow as follows:
[image]

hope this can help.

HobbytronicsPK, 2019-05-13:
Have a look on to [Ghost / Random Switching on Sonoff Devices (MQTT Retained)](https://blog.hobbytronics.pk/sonoff-tasmota-tips-tricks/#Ghost-Random-Switching)