arendst/Tasmota #2197: Sonoff 4ch, wall rocker switches

coffman34, 2018-03-19:
Using a Sonoff 4ch. I have Tasmota flashed and working. My issue is I am using wall switches, that are soldered on the back of the push buttons on the device. I need to change these from momentary to regular rocker switches. I have tried setting SwitchModeN 1, but they still act as momentary buttons.

I think I need to define these as switches, in my Configuration > Module, I only see GPIO1, GPIO2, GPIO3. I'm a little confused on where to go from here.

Frogmore42, 2018-03-19:
look at https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

coffman34, 2018-03-20:
I looked at that page before. As stated. I tried to set the SwitchMode to 1, to 7, nothing changes. My next step was to make sure that it's actually using the GPIO pins as a switch, however, in the Sonoff 4ch module config page, there is only GPIO1,2,3. No 0,9,10,14 like I believe I need to configure.

coffman34, 2018-03-20:
I believe I got it figured out. I had to set it as a 4ch Pro. Then the SwitchMode commands worked correctly. For some reason setting it to 4ch only wasn't accepting the SwitchMode commands.

closed 2018-03-20 by coffman34