arendst/Tasmota #2198: Double Recognized Sonoffs with Hue Emulation when Echo Dot 2. Generation is connected

motorrad, 2018-03-19:
Make sure these boxes are checked before submitting your issue - Thank you!

Im using 2 X Echo Show 3 X Echos 1 X Sonos One 2 X Echo Spot and after Connecting an Echo Dot 2.nd Generation the Philips Hue Emulation is discovered two Times for every Sonoff device :
First as : Royal Philips Electronics intelligentes Gerät 
Second as : Philips Hue
Belkin Emulation remains ok.
I am using the v5.12.0 Firmware.
When I disconnect the Echo Dot there are only Royal Philips Electronics intelligentes Gerät devices which will be found.

motorrad, 2018-03-20:
THX@arendst  for Setting help wanted Label :-)

vw2audi, 2018-05-02:
Hi i have exactly the same problem. Any sollution?

closed 2018-07-01 by stale[bot]