arendst/Tasmota #2199: Is it normal to have many (10s to 100s) of "problems" when compiling with PlatformIO?

Frogmore42, 2018-03-19:
Make sure these boxes are checked before submitting your issue - Thank you!

- [NA] What hardware you are using
- [NA] Provide the output of command ``status 0``
- [NA] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [NA] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

I have been using Tasmota for months and it has generally been great and a real timesaver.  I am currently working on creating a new sensor function to gather information from my water meter (it is not a standard pulse output, but rather a direct reading register).

I am new to PlatformIO, but have used several different IDEs (including Visual Studio, IAR and Keil) for software/firmware development.  While I have been successful in getting things to compile (and even work) it is troubling that PlatformIO seems confused by the Tasmota code.

1. When code is inside of an #ifdef, VSCode (or PlatformIO) does not seem to know if the code is active or not, seems like it is not, since it is greyed out.

1. It seems like it can't find definitions for some things, though they are really defined in the code (somewhere) and it builds it just fine.

1. There are literally hundreds of flagged issues, which seem like they should be fatal, but the code still compiles and runs (and this is before I have made any changes to it).

1. It seems that VSCode or PlatformIO take awhile to catch up to changes when spread across a few files (like a .h and .cpp).  It does catch up eventually.

1. When adding a new .ino to implement a new sensor, I had a really hard time getting PlatformIO to notice there was a new file there and to compile it.  I deliberately put a syntax error in it so that it would fail, but it did not, so I am pretty sure it was not compiling it.  I moved it to a library, which did not work and then moved it back and then it started seeing the new file (.ino) and compiling it.

So, are other people seeing these kinds of issues with PlatformIO?
Is it that the Tasmota code really has all these problems, or is something misconfigured for PlatformIO or for VSCode?

The only reason I am asking is that it is much easier to find the issues I introduce if I know the code before was clean and had no errors or warnings.

Frogmore42, 2018-03-20:
What does the invalid label mean?

Frogmore42, 2018-03-20:
So I looked it up and Git says this about invalid
Indicates that an issue or pull request is no longer relevant
That does not seem to apply.  I am fully willing to believe there is something wrong on my end and am willing to figure it out.  But, if everyone is seeing the same thing, there probably isn't an issue on my end.

curzon01, 2018-03-20:
Issue lists for projects on Github are often used like a forum, but it should be issues related and only related to the project. Most people (even beginners) do not differentiate between this, everyone place their questions in this list related to the whole equipment they are using with Tasmota.

About your issue is depending on Tasmota because it happens for you with Tasmota code but when I read it, the questions and problems you have are of common nature related to VSC and PlatformIO. Could be that's the reason why it is flag as `invalid`

Frogmore42, 2018-03-20:
I agree that many people use the issues list as a forum.  I am sure that is because there is no forum.  Many people also never come back to close the issue after it is presumably resolved, since they answer none of the questions asked of them in response to their query.  I am a firm believer in having a clean issues list and I always check it first to see if the issue is new.

Since you responded, do you use PlatformIO and/or VSC?  

Tasmota is a large and complex software project.  I am getting a better feel for how it is constructed, but it is probably too large for the "simplifications" that using Arduino INO files enables.  It is possible that it what is causing the issues I am seeing.  Or, perhaps my setup is misconfigured.  I wouldn't doubt that at all.

arendst, 2018-03-20:
@Frogmore42 I did add the label invalid as this is a general VSC/Platformio issue. Tasmota on VSC should run fine. I also see problems on the problems tab but the nice thing is they come and go even if I do nothing so I expect VSC and platformio still have to iron out some things.

Especially when switching from 1.5.0 to 1.6.0 libraries and back you'll see compile errors. Just ignore them and retry the compile after you have selected the clean action first.

I also receive regular Intellisense errors due to several Python installs on my machines. This seems to be a Window issue to be resolved in the next major windows update!

As long as the compilation runs fine you're good to go. 

We're working on the forum (see #2059). The idea is to move all Help Wanted issues to the forum using a script. Until then I scam the issue quick and add a label to it to automate issue move.

Frogmore42, 2018-03-20:
Okay, I do remember seeing something about a forum.  I agree it will be great to have that for these types of questions.  Thanks for letting me know that you see the issues with VSC and PlatformIO too.

closed 2018-03-20 by Frogmore42