arendst/Tasmota #2200: Esp8285 memory issue - Sonoff Dev

iotomic, 2018-03-19:
I've been attempting to flash a bunch of Sonoff Dev boards and get stuck at the upload step.

I've reviewed the previous threads on this topic but still get the issue that it thinks the board only contains 512k. Have attached screenshot of platform.io and error.

Am I missing something? I have a few of these boards and I'm keen to use them for simple temp->MQTT telemetry.

[image]
[image]

Jason2866, 2018-03-20:
What happens iy you try:  
command "Clean" and then only command "Build"
Does it build the firmware. If yes it is a upload problem. If no it is a Platform Atom.IO problem in your installation

arendst, 2018-03-20:
You missed reading the wiki regarding how to use the package. Your platformio.ini board type is wrong. See package for correct and only supported board type.

kajmaj, 2018-03-20:
What is the right way to enter flashing mode?
My Sonoff DEV test piece just arrived, but steps described in wiki does not work for me. (press and hold E-FW button and short press of Reset button)

Jason2866, 2018-03-20:
Just press and hold E-FW before powering device. Reset button is not needed.
Now you should in flash mode. Now connect usb serial adapter (or if you power the device via adapter hold E-FW button) to usb port of pc. Now you can upload the firmware

kajmaj, 2018-03-20:
How do I recognise, that I am in Flashing mode?
I did as you described, red LED is on. Is it OK?

Jason2866, 2018-03-20:
Red led is that device is powered, not more.
Simply try to flash. There is no indicator that you are in flash mode.
If it not works you can try, reset flash mode from NodeMcu. The dev board has a similar Flash/Reset logic on board as the NodeMcu. For that you have to delete the ";" (if you use Atom.Io!) In platformio.ini
;upload_resetmethod = nodemcu
With that you simply connect your dev board with usb cable and do the flash process with command "Upload". Thats it

kajmaj, 2018-03-20:
Done !! I flashed it via FTDI adapter.
You are the magician :) Thank you again.