arendst/Tasmota #2201: getting errors when compiling a sketch

wolph42, 2018-03-19:
Hi, completely new at this and I need some help. Im trying to compile one of the sketches, e.g: compiling: xdrv_07_home_assistant.ino gives the following errors:

----------------------------
Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 1M (64K SPIFFS), 2, v2 Lower Memory, Disabled, SSL+TLS_MEM, Only Sketch, 115200"

Archiving built core (caching) in: C:\Users\ROLPHS~1\AppData\Local\Temp\arduino_cache_931258\core\core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_1M64,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_SSLTLS_MEM,FlashErase_none,UploadSpeed_115200_007cfbad64b6ffec161ff98b5041ce3a.a
C:\Users\ROLPHS~1\AppData\Local\Temp\arduino_build_87823/arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'

C:\Users\ROLPHS~1\AppData\Local\Temp\arduino_build_87823/arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'

C:\Users\ROLPHS~1\AppData\Local\Temp\arduino_build_87823/arduino.ar(core_esp8266_main.cpp.o): In function `loop_wrapper':

C:\Users\Rolph Segers\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp:111: undefined reference to `setup'

C:\Users\Rolph Segers\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp:111: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

--------------------
Compiling Arduino IDE on windows 10. Concerns a Sonoff TH10

any clue on what i'm doing wrong? 

pls note that i want it eventually to work with blynk. 

thank you!

wolph42, 2018-03-20:
As ik completely new to this, could you please elaborate on why my question was invalid? Thank you!

mscreations, 2018-03-21:
In Arduino IDE, open the sonoff.ino file. That is the file you should be compiling, not any of the others.

wolph42, 2018-03-22:
thank you, it still required quite a bit of tinkering to get this running under arduino IDE but finally I got it to compile.

closed 2018-03-22 by wolph42