arendst/Tasmota #2204: The green LED doesn't flash after upload Tasmota

dzinyit, 2018-03-20:
I was trying to flash Tasmota to my Sonoff Basic, however after upload finished, the green LED just didn't flash as shown in tutorials on youtube. I was thinking could it possible the hardware is somehow not working properly when I soldered. I didn't plug in AC power in case it's going to be damaged permanently. Here's what's shown on Arduino IDE, any help is appreciated. 

> Archiving built core (caching) in: C:\Users\DzinYit\AppData\Local\Temp\arduino_cache_775984\core\core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_nodemcu,CrystalFreq_26,FlashFreq_40,FlashMode_dout,FlashSize_1M0,LwIPVariant_Prebuilt,Debug_Disabled,DebugLevel_None____,UploadSpeed_115200_3e4552dc23806ecfee95d46bf6d12975.a
> Sketch uses 502772 bytes (49%) of program storage space. Maximum is 1023984 bytes.
> Global variables use 51340 bytes (62%) of dynamic memory, leaving 30580 bytes for local variables. Maximum is 81920 bytes.
> Uploading 506928 bytes from C:\Users\DzinYit\AppData\Local\Temp\arduino_build_539725/sonoff.ino.bin to flash at 0x00000000
> ................................................................................ [ 16% ]
> ................................................................................ [ 32% ]
> ................................................................................ [ 48% ]
> ................................................................................ [ 64% ]
> ................................................................................ [ 80% ]
> ................................................................................ [ 96% ]
> ................                                                                 [ 100% ]
>

arendst, 2018-03-20:
Perhaps it doesn't need to flash?

dzinyit, 2018-03-20:
> Perhaps it doesn't need to flash?

Could be possible. I hadn't paid attention to the LED the first time I flashed, and then flashed a second time.

closed 2018-03-23 by dzinyit