arendst/Tasmota #2206: Cannot get a AP on by Dual R2

NiekN, 2018-03-20:
i used the bin file, just like the sonoff basic, and was completed without error
esptool.exe -vv -cb 115200 -cp COM10 -ca 0x00000 -bz 1M -bm dout -cf sonoff.bin

but at my sonoff basic, when i press 3x short the butten i get the AP
but at dual it does not work ?   the blue LED blinks about 33x then continu on then 33x blinking etc
The green LED is continu on 

need help??

Jason2866, 2018-03-20:
I took a look in sonoff_template.h I didnt see a button function for the dual.
May i overlooked something, but this would describe your problem why it not works

NiekN, 2018-03-20:
But how can i get it work?

Jason2866, 2018-03-20:
Compile your own versions with settings for WLAN in user_config.h.
The sonoff Dual will connect with your WLAN and you configure it for your needs

NiekN, 2018-03-20:
where and how to compile ? i tried arduino.cc  followed the steps but i get an error 
https://ru-clip.com/video/pGbDFsNMOkw/sonoff-dual-with-tasmota-firmware.html

#error "MQTT_MAX_PACKET_SIZE is too small in libraries/PubSubClient/src/PubSubClient.h, increase it to at least 1000"

Jason2866, 2018-03-20:
Do what the error message tells you.
If you install the Ide Platform Atom.io it is easier no "hand copying and editing files"
Look in the wiki it is described

NiekN, 2018-03-21:
I did increase it to 1000 but then i get the next error
**Error compiling for board Generic ESP8266 Module.** 

but i will try ATOM first , it looks like the wiki is explaining step by step

NiekN, 2018-03-21:
@arendst 
So i have to used BUTTON 0 (= GPIO0) connected to GND. Thats actually my button , like at the sonoff basic.

arendst, 2018-03-21:
Yep.

NiekN, 2018-03-21:
Thx it works

closed 2018-03-21 by NiekN