arendst/Tasmota #2207: SHT3X-0x45 Temperature and Humidity labels

RaymondMouthaan, 2018-03-20:
Make sure these boxes are checked before submitting your issue - Thank you!

- [x] What hardware you are using: Wemos D1 Mini Pro + SHT30

Are these labels for indicating temperature and humidity new? The "-0x45" is seen since tasmota version 5.12.0g. Screenshot below is version 5.12.0h

[image]

arendst, 2018-03-20:
Yep. it now supports two of them. The dash was there all the time on other sensors so I suggest to google around for a javascript solution.

RaymondMouthaan, 2018-03-21:
Theo that was just the post i was checking haha .. 

I've fixed it using the bracket notation instead of the dot notation. 

So my function in node-red looks like this:
 
```
// Prepare SHT3X-0x45 Temperature and Humidity for InfluxDB
msg.payload = msg.payload["SHT3X-0x45"];
msg.measurement = msg.topic.match(".*/(.*)/.*")[1];
return msg;
```

Thanks for pointing me in a different direction :-)

arendst, 2018-03-21:
Glad I could help ;;;---)))

closed 2018-03-21 by arendst

mrteausa, 2018-12-11:
Hello, I put this value_template: '{{ value_json["SHT30-0x45"].Temperature }}' on the configuration.yaml in HA it does not get the reading why??

cmccambridge, 2019-01-24:
@mrteausa - There's a typo in the template you quoted. The correct name for the JSON field is `SHT3X-0x45`, as in: `{{ value_json['SHT3X-0x45'].Temperature }}`.