arendst/Tasmota #2208: Add all friendly names to the status payload

mizrachiran, 2018-03-20:
Hi,

This is a feature request.
I would love to see all friendly names in the status payload, currently only the first is printed out.
This can be useful for third party integration.

Thanks!

```
  if ((0 == payload) || (99 == payload)) {
    snprintf_P(mqtt_data, sizeof(mqtt_data), PSTR("{\"" D_CMND_STATUS "\":{\"" D_CMND_MODULE "\":%d,\"" D_CMND_FRIENDLYNAME "\":\"%s\",\"" D_CMND_TOPIC "\":\"%s\",\"" D_CMND_BUTTONTOPIC "\":\"%s\",\"" D_CMND_POWER "\":%d,\"" D_CMND_POWERONSTATE "\":%d,\"" D_CMND_LEDSTATE "\":%d,\"" D_CMND_SAVEDATA "\":%d,\"" D_JSON_SAVESTATE "\":%d,\"" D_CMND_BUTTONRETAIN "\":%d,\"" D_CMND_POWERRETAIN "\":%d}}"),
      Settings.module +1, Settings.friendlyname[0], mqtt_topic, Settings.button_topic, power, Settings.poweronstate, Settings.ledstate, Settings.save_data, Settings.flag.save_state, Settings.flag.mqtt_button_retain, Settings.flag.mqtt_power_retain);
    MqttPublishPrefixTopic_P(option, PSTR(D_CMND_STATUS));
  }
```

Jason2866, 2018-03-20:
No offend! What is easier, for integration if friendly name is in?
For my point of view it makes parsing the payload more complicated, when redundant data is in.

mizrachiran, 2018-03-20:
It will help knowing the name of each device when integrating with other systems. 

I can’t understand how it makes parsing harder...

Jason2866, 2018-03-20:
I didnt understand it correct. Read it again. Now it is clear. Has no affect in parsing. ..
But i still do not see the benefit. If the friendly name is needed command "status" and there it is.

mizrachiran, 2018-03-20:
This is true. But only for the first switch / relay. 
Each device you connect to tasmota has it’s own friendly name. 

See in the code snippet i attached how we take the first out of the array.

Jason2866, 2018-03-20:
Lets see what @arendst think about it 🤔

arendst, 2018-03-21:
Have a look at HomeAssistant Mqtt discovery in Tasmota. It provide the friendlynames on startup in a retained message to be used by Hass or Domoticz. 

You might want to jump on the same bandwagon for your third party integration.

mizrachiran, 2018-03-21:
thanks @arendst 
The thing is that i'm trying to integrate sonoff tasmota devices into an iOS app that doesn't work with MQTT. 
It uses mDNS to discover devices and making http calls to get all information and control the devices.
Is there an option to get all friendly names by http call?

Thanks!
Ran

arendst, 2018-03-21:
Currently the only way is to execute commands like ``friendlyname2``.

I will implement your request as an array of friendlynames in the status command. Expect something like 
```
... ,"FriendlyName":["DualR2 1","DualR2 2"], ...
```

mizrachiran, 2018-03-21:
@arendst 
This is great! 
Thank you so much!

closed 2018-03-21 by arendst