arendst/Tasmota #2209: SONOFF Basic goes Offline and back randomly

akasma74, 2018-03-20:
I use SONOFF Basic flashed with Tasmota 5.12.0g. It is connected to TP-LINK TL-WR702N in AP mode. 
It is integrated with Home Assistant 0.64.3 (Raspberry Pi 3):
In configuration.yaml:
`switch:
- platform: mqtt
    name: "Main siren"
    state_topic: "home/siren/RESULT"
    value_template: '{{ value_json["POWER"] }}'
    command_topic: "home/siren/cmnd/POWER"
    availability_topic: "home/siren/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true
`
and in automations.yaml:
- id: main_siren_unavailable
  alias: '[int] notify when main siren becomes unavailable'
  trigger:
  - platform: mqtt
    topic: !secret main_siren_availability_topic
    payload: !secret main_siren_unavailable
  action:
  - service: persistent_notification.create
    data_template:
      title: '[Alarm] Fault: main siren is unavailable!'
      message: "Main siren is unavailable at {{ now().strftime('%H:%M on %d %b %Y') }}"

In output of mosquitto_sub -h 192.168.0.34 -v -t "#" I can see a lot of messages similar to these:

`home/siren/STATE {"Time":"2018-03-20T21:41:34","Uptime":"0T04:40:44","Vcc":3.465,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":56,"APMac":"A0:F3:C1:A9:13:66"}}
home/siren/LWT Offline
home/siren/LWT Online
home/siren/cmnd/POWER (null)
home/siren/RESULT {"POWER":"OFF"}
home/siren/POWER OFF
home/siren/LWT Offline
home/siren/LWT Online
home/siren/cmnd/POWER (null)
home/siren/RESULT {"POWER":"OFF"}
home/siren/POWER OFF
home/siren/LWT Offline
home/siren/LWT Online
home/siren/cmnd/POWER (null)
home/siren/RESULT {"POWER":"OFF"}
home/siren/POWER OFF
home/siren/STATE {"Time":"2018-03-20T21:46:40","Uptime":"0T04:45:50","Vcc":3.462,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":56,"APMac":"A0:F3:C1:A9:13:66"}}
home/siren/LWT Offline
home/siren/LWT Online
home/siren/cmnd/POWER (null)
home/siren/RESULT {"POWER":"OFF"}
home/siren/POWER OFF
home/siren/LWT Offline
home/siren/LWT Online
home/siren/cmnd/POWER (null)
home/siren/RESULT {"POWER":"OFF"}
home/siren/POWER OFF
home/siren/STATE {"Time":"2018-03-20T21:51:43","Uptime":"0T04:50:53","Vcc":3.466,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":54,"APMac":"A0:F3:C1:A9:13:66"}}
home/siren/LWT Offline
home/siren/LWT Online
home/siren/cmnd/POWER (null)
home/siren/RESULT {"POWER":"OFF"}
home/siren/POWER OFF
home/siren/STATE {"Time":"2018-03-20T22:01:46","Uptime":"0T05:00:56","Vcc":3.463,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":54,"APMac":"A0:F3:C1:A9:13:66"}}
home/siren/UPTIME {"Time":"2018-03-20T22:02:00","Uptime":"0T05:01:10"}
home/siren/STATE {"Time":"2018-03-20T22:06:46","Uptime":"0T05:05:56","Vcc":3.463,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":54,"APMac":"A0:F3:C1:A9:13:66"}}
`

The output of command ``status 0``:

`home/siren/cmnd/status 0
home/siren/STATUS {"Status":{"Module":1,"FriendlyName":"siren","Topic":"siren","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
home/siren/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T05:26:05","StartupUTC":"2018-03-20T16:00:50","Sleep":0,"BootCount":24,"SaveCount":96,"SaveAddress":"F4000"}}
home/siren/STATUS2 {"StatusFWR":{"Version":"5.12.0g","BuildDateTime":"2018-03-20T12:14:26","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
home/siren/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId1":"Level2","SSId2":"","TelePeriod":300,"SetOption":"00000009"}}
home/siren/STATUS4 {"StatusMEM":{"ProgramSize":497,"Free":504,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
home/siren/STATUS5 {"StatusNET":{"Hostname":"siren-6348","IPAddress":"192.168.0.37","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"194.168.4.100","Mac":"60:01:94:69:38:CC","Webserver":2,"WifiConfig":3}}
home/siren/STATUS6 {"StatusMQT":{"MqttHost":"192.168.0.34","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6938CC","MqttUser":"HA","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
home/siren/STATUS7 {"StatusTIM":{"UTC":"Tue Mar 20 21:26:55 2018","Local":"Tue Mar 20 22:26:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
home/siren/STATUS10 {"StatusSNS":{"Time":"2018-03-20T22:26:55"}}
home/siren/STATUS11 {"StatusSTS":{"Time":"2018-03-20T22:26:55","Uptime":"0T05:26:05","Vcc":3.464,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":52,"APMac":"A0:F3:C1:A9:13:66"}}}
home/siren/STATE {"Time":"2018-03-20T22:27:01","Uptime":"0T05:26:11","Vcc":3.462,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":54,"APMac":"A0:F3:C1:A9:13:66"}}
`

In the last 3 hours it went offline at least 18 times. When I press its button, it tortes on and off, i.e it's not completely dead, and I can control it from HA's fronted.
However, I created the automation main_siren_unavailable to be aware if the relay if offline as I'm trying to build a burglar alarm, and such thing without a working siren (which is powered through SONOFF Basic) forth nothing.

What does it all mean and how to fix or debug it? I moved from Espurna because of availability issues, but with different board and different firmware I still have similar issues.

ionciubotaru, 2018-03-21:
Check the wifi signal strength. This is the most common cause.

akasma74, 2018-03-21:
Well, the thing is in almost clear sight from the access point, no more than 5 metres unobstructed.
This AP serves several more devices, none of them seem to cause any overload or suffer from poor signal.
How can I check if it's because of the signal strength?

arendst, 2018-03-21:
If I see yaml it must be tagged as Help wanted ;-)

akasma74, 2018-03-21:
Sorry Theo, that was my very first post here and I could not find where to apply tags.. :\

arendst, 2018-03-21:
No problem. It was kind of meant sarcastic from me. Forget it. I really have to refrain from bashing Hass but it's so hard...

BTW. As far as I know a user cannot add tags so that's not your problem either :-)

akasma74, 2018-03-21:
I see.
Hass is my very first home automation platform, and so far I'm pleased with it because it allows me to achieve good results quickly and without diving into complex matters.
I started messing with all that automation stuff in a bid to build a burglar alarm based on HA (RPi 3B) and connected to SONOFF RF bridge (Espurna) with a bunch of SONOFF PIRs and door switches (DW1s).
Currently I'm hoping to extend it by adding some RF smoke detectors, RF temperature&humidity sensors and RF wall switches.
It's clearly grows bigger than just a burglar alarm, but I can always leave the alarm part lone and deploy anything else on another RPi/PC/whatever is it's too much for one.
What would you recommend as an alternative to Hass?
And why do your criticise it?

arendst, 2018-03-21:
I'm using Domoticz because it allows to make configuration changes without taking down the HA environment where Hass has to be restarted for every single change in yaml config file. I have Domoticz up and running for months on Raspi while still easily changing configurations either in the Gui or in scripts.

Hass also heavily relies in MQTT retain messages leading the a host of issues here as people just don't get why their device restarts at strange moments due to the MQTT broker re-sending retained messages.

Retaining is not needed as Tasmota saves the power state every time it is changed so even after a power drop it changes the state to the last known state or, depending on user settings to another known state.

But as I wrote before I'll start counting down from ten before ever thinking of bashing Hass again ;-)

mr-sneezy, 2018-03-21:
Theo, if one where to be interested in Domoticz on Raspi, what forums are the best for looking at for info ?

akasma74, 2018-03-21:
Theo, thanks for pointing out Hass's weaknesses.
I agree it's not ideal to restart HA on every change.
And I presume I understand what you are talking about regarding retained messages ;)
Yesterday when I flashed Tasmota to the SONOFF Basic and tested it, everything worked fine except of one thing. I created a group of switches and included that Basic one and dead SONOFF SV into it. So it showed SV Unavailable and Basic as normal. But when pressed the button on Basic so it turned ON and then disconnected from power, waited until the whole group became Unavailable and powered Basic back again, it came back to ON, and then after a few seconds was turned OFF. I presume because Unavailable group cannot be ON, and when it became Available, its state was OFF and HA just sent MQTT OFF message to both devices. Oh well..

arendst, 2018-03-21:
I would say google Domoticz but hey look here http://www.domoticz.com/ Raspi ready downloads and forum.

The original issue might be solved by completely erase and re-flash the unit.

I see something similar issues these days and switching broker does not help. Switching to another AP seems to help so I suspect an overloaded AP where I have more than 20 devices connected to. Also reflashing between esp/arduino libraries 2.3.0 and 2.4.0 sometimes solves it. I can't quit point the finger at it...

akasma74, 2018-03-21:
> The original issue might be solved by completely erase and re-flash the unit.

will esptool.py erase_flash do the job?
don't think my AP is overloaded as it serves only RF bridge and SONOFF Basic at the moment, and the RF bridge works fine in terms of being available. But they are different devices, aren't they? :\

arendst, 2018-03-21:
Yep. See wiki.

akasma74, 2018-03-21:
Can't see anything on wiki, could you post a link?

arendst, 2018-03-21:
You're teasing me. I thought a name like "Troubleshooting" would be selfexplainatory ... https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting

akasma74, 2018-03-21:
I'm not ;)
The wiki is big, and I checked tips, specific articles but not Troubleshooting.
Maybe it's a good idea a to move esptool section into Tools and leave a link to it from Troubleshooting?

arendst, 2018-03-21:
Just looked in https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool. The info is there already....

Have a nice day!

Barnabybear, 2018-03-26:
Hi, have a look at #2149

akasma74, 2018-04-10:
@Barnabybear thanks for theta, but it does not look like my case.
Today my SONOFF Basic (which duty is to switch on and off pretty loud alarm siren) did the following:
home/siren/STATE {"Time":"2018-04-10T17:55:39","Uptime":"13T01:59:02","POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":38,"APMac":"E8:94:F6:86:4D:96"}}
then
home/siren/LWT Offline
home/siren/LWT Offline
home/siren/LWT Online
home/siren/cmnd/POWER (null)
home/siren/RESULT {"POWER":"ON"}
home/siren/POWER ON
home/siren/RESULT {"POWER":"OFF"}
home/siren/POWER OFF
home/siren/STATE {"Time":"2018-04-10T18:00:58","Uptime":"13T02:04:21","POWER":"OFF","Wifi":{"AP":1,"SSId":"Level2","RSSI":44,"APMac":"E8:94:F6:86:4D:96"}}
home/siren/UPTIME {"Time":"2018-04-10T18:02:00","Uptime":"13T02:05:23"}

I.e at some point I received a notification that my siren became unavailable (but remained on, no reboots), and almost immediately it switched  back on so I had to go and press the button to switch it off.
It was the second time when it became unavailable and upon coming back turned itself on.
Any ideas what it might be?