arendst/Tasmota #2210: sonoff:42: error: #error Please use stable lwIP v1.4

PETEIOT, 2018-03-20:
I continue to get this error even if I only verify the code, I have changed the packet size in PubSubClient 1000 I've tried different boards, restarting, re-downloading any idea? 

Running windows, 

sonoff:42: error: #error Please use stable lwIP v1.4

   #error Please use stable lwIP v1.4

    ^

exit status 1
#error Please use stable lwIP v1.4

mscreations, 2018-03-21:
Make sure you setup Arduino IDE similar to below (at least for the development branch):

IF you are on the master branch, then choose one of the v1.4 options. 

[image]

PETEIOT, 2018-03-21:
OK so I was thinking there was a problem with my options  I don’t have those same options I’m missing the flash mode and the one that you’re referring to and if you others is there something I need to turn on?

mscreations, 2018-03-21:
I forgot to mention that you also should go into Board Manager (Board submenu) and make sure esp8266 version is 2.4.1. I've had the most success with this particular version and setup.

[image]

PETEIOT, 2018-03-21:
That was it thanks mscreations!

closed 2018-03-21 by PETEIOT

Sovatna, 2018-08-13:
Thanks @mscreations you also solve my doubt.