arendst/Tasmota #2212: Add blade counter in shutter mode

renne, 2018-03-21:
Rating the current shutter position is quite complicated. The runtime of the shutter needs to be clocked exactly and exponentially multiplied with the dynamic radius of the shutter winding on the shaft.

I suggest to mount pieces of adhesive magnetic stripe on each shutter blade and a hall sensor connected to a GPIO in the shutter box. That way the position of the shutter can be calculated easily:

**pos = 100% / blades * counter**

renne, 2018-06-06:
As the WLAN connection can fail I suggest to implement this directly in Tasmota.

stefanbode, 2018-07-02:
I have now addressed this issue in my mod of Tasmota. You will find it under stefanbode. Functionality is to open close and set to any value between 0 and 100. All calculation and Relay control is done in the firmware. Please read the wiki carefully how to set it up and let me know if you need support.

ascillato2, 2018-08-25:
Hi,

Continuing this issue at: #3137

closed 2018-08-25 by ascillato2