arendst/Tasmota #2213: error while compiling

rkn45, 2018-03-21:
I am using Sonoff dual R2

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v1.4 Compile from source, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mikro\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mikro\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,led=2,LwIPVariant=OpenSource,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -vid-pid=0X2341_0X0042 -ide-version=10805 -build-path C:\Users\mikro\AppData\Local\Temp\arduino_build_761450 -warnings=none -build-cache C:\Users\mikro\AppData\Local\Temp\arduino_cache_239241 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -verbose F:\Arduino\sonoff\sonoff.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\mikro\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\mikro\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries F:\Arduino\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,led=2,LwIPVariant=OpenSource,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -vid-pid=0X2341_0X0042 -ide-version=10805 -build-path C:\Users\mikro\AppData\Local\Temp\arduino_build_761450 -warnings=none -build-cache C:\Users\mikro\AppData\Local\Temp\arduino_cache_239241 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.13 -verbose F:\Arduino\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1
Using core 'esp8266' from platform in folder: C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1
Detecting libraries used...
"C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\mikro\AppData\Local\Temp\arduino_build_761450/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "C:\Users\mikro\AppData\Local\Temp\arduino_build_761450\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\mikro\AppData\Local\Temp\arduino_build_761450/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "-IF:\Arduino\libraries\PubSubClient-2.6.09\src" "C:\Users\mikro\AppData\Local\Temp\arduino_build_761450\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\mikro\AppData\Local\Temp\arduino_build_761450/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "-IF:\Arduino\libraries\PubSubClient-2.6.09\src" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\Ticker" "C:\Users\mikro\AppData\Local\Temp\arduino_build_761450\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\mikro\AppData\Local\Temp\arduino_build_761450/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "-IF:\Arduino\libraries\PubSubClient-2.6.09\src" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\Ticker" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src" "C:\Users\mikro\AppData\Local\Temp\arduino_build_761450\sketch\sonoff.ino.cpp" -o "nul"
"C:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/in [... cut ...] ro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\Wire" "-IF:\Arduino\libraries\IRremoteESP8266-2.2.1.02\src" "-IF:\Arduino\libraries\TasmotaSerial-1.1.0\src" "-IF:\Arduino\libraries\NeoPixelBus-2.2.9\src" "-IC:\Users\mikro\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\SPI" "C:\Users\mikro\AppData\Local\Temp\arduino_build_761450\sketch\sonoff.ino.cpp" -o "C:\Users\mikro\AppData\Local\Temp\arduino_build_761450\preproc\ctags_target_for_gcc_min

arendst, 2018-03-21:
Nice! Love the amount of data you provide. I'll send it to Cambridge Analytica...

Bolukan, 2018-03-23:
Maybe check your user_config.h

closed 2018-07-10 by ascillato2