arendst/Tasmota #2214: Today's recordings get reset

earth08, 2018-03-21:
Dear what does this means,
I have physical switch before sonoff power
I have installed 5.12f
When I switchoff physical switch, the counter gets reset,
Please guide me,
[image]

Also in morning from 12:30 am to around 4:30 am or 5:00 am, no readings are recorded even when my air-conditioner is on,
[image]

arendst, 2018-03-21:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``

earth08, 2018-03-21:
Dear 
Sorry
I use Sonoff Pow
and output of satus o
20:24:18 CMD: status0
20:24:18 MQT: stat/ps101/STATUS = {"Status":{"Module":6,"FriendlyName":"powerswitch101","Topic":"ps101","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:24:21 MQT: tele/ps101/STATE = {"Time":"2018-03-21T20:24:21","Uptime":"0T00:00:30","Vcc":3.473,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Jay Ambe 1","RSSI":54,"APMac":"34:CE:00:59:AC:99"}}
20:24:21 MQT: tele/ps101/SENSOR = {"Time":"2018-03-21T20:24:21","ENERGY":{"Total":6.153,"Yesterday":1.505,"Today":2.608,"Period":0,"Power":0,"Factor":0.00,"Voltage":0,"Current":0.000}}

earth08, 2018-03-22:
Dear this is todays
i have cleared all data before 9:00 AM
and this is what i get with status0
08:41:56 CMD: status0
08:41:56 RSL: stat/ps101/STATUS = {"Status":{"Module":6,"FriendlyName":"powerswitch101","Topic":"ps101","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

todays got reseted and becomes zero
as proof i am attaching 
grafana graph

[image]

HomeAssistant Graph

[image]

Frogmore42, 2018-03-22:
When you abruptly remove power from a POW device it will lose data as it only saves to flash on occasion to minimize flash writes.  It saves data in RTC ram, which will forget stuff if power is off for too long.

earth08, 2018-03-22:
What should be done to correct this,
If we start/power on what should be done to get details from mqtt server?
Is there anyway we can get correct details even after powering off Sonoff power for a long time

curzon01, 2018-03-22:
Do not power off your Sonoff Pow

arendst, 2018-03-22:
Or perform the restart command before powering off the unit. The restart will save volatile data to be available when power is re-applied.

earth08, 2018-03-22:
Dear thanks for your guidance,
But this happens unknowingly by kids.
Is there anyway to get information  from mqtt server so that I can get atleast the last recorded value

arendst, 2018-03-22:
Nope.

The way I use it is having a HA like Domoticz which receives every TelePeriod of 5 minutes the delta energy usage. Domoticz then stores the information in it's database and provides nice graphics like:

[image]

Good enough for me.

earth08, 2018-03-22:
I am using grafana and influxdb but once the switch is poweroff, today's reading get's reset,
What is best, pls recommend
How to use domoticz,
What app have u used for graph

arendst, 2018-03-22:
Domoticz is a one size fits all. The graphics is included! See http://www.domoticz.com/

closed 2018-03-22 by earth08

earth08, 2018-03-23:
Dear,
today i found that even with constant power on, whenever i restart my sonoff pow with webbrowser, it toggles to off,
what should be done to retain its position.
Please guide
10:16:48 CMD: status 0
10:16:48 MQT: stat/ps101/STATUS = {"Status":{"Module":1,"FriendlyName":"PowerSwitch101","Topic":"ps101","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:16:48 MQT: stat/ps101/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:37","StartupUTC":"2018-03-23T04:46:11","Sleep":0,"BootCount":6,"SaveCount":15,"SaveAddress":"F5000"}}
10:16:48 MQT: stat/ps101/STATUS2 = {"StatusFWR":{"Version":"5.12.0f","BuildDateTime":"2018-03-22T14:35:46","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
10:16:48 MQT: stat/ps101/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"192.168.11.33","LogPort":514,"SSId1":"Jay Ambe","SSId2":"Jay Ambe 1","TelePeriod":300,"SetOption":"00000009"}}
10:16:48 MQT: stat/ps101/STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":484,"Heap":18,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
10:16:48 MQT: stat/ps101/STATUS5 = {"StatusNET":{"Hostname":"ps101-7360","IPAddress":"192.168.11.101","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"2C:3A:E8:07:1C:C0","Webserver":2,"WifiConfig":3}}
10:16:48 MQT: stat/ps101/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.11.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_071CC0","MqttUser":"kiran","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:16:48 MQT: stat/ps101/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 23 04:46:48 2018","Local":"Fri Mar 23 10:16:48 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
10:16:48 MQT: stat/ps101/STATUS10 = {"StatusSNS":{"Time":"2018-03-23T10:16:48"}}
10:16:48 MQT: stat/ps101/STATUS11 = {"StatusSTS":{"Time":"2018-03-23T10:16:48","Uptime":"0T00:00:37","Vcc":3.473,"POWER":"ON","Wifi":{"AP":1,"SSId":"Jay Ambe","RSSI":72,"APMac":"08:BD:43:FE:43:FD"}}}

curzon01, 2018-03-23:
1. Unfortunately you do not use the Tasmota suggested topic which defaults are different for `cmnd` and `stat`. So your topic `ps101` is the same for cmnd and stat messages.

2. Your are sure that you do not have a retained message stored on your MQTT server which leads your Sonoff Pow to off? Retained messages are very often the reason for unwanted issues - I hate retained if it's unsolicitous used.
First check your MQTT broker about retained POWER commands for your Sonoff Pow.

earth08, 2018-03-23:
Dear Curzon01,
When you say suggested tasmota topic,
Please explain,
I have now switched my Sonoff power MTT topic to pswitch101.
As suggested by you,
How to get tasmota suggested topic,

curzon01, 2018-03-23:
? I did not suggest to change your topic. My mistake was I thought the "Topic" above is your FullTopic, but it isn't. What is your FullTopic (enter `FullTopic` on web console)?

But before continue: Check 2. on my comment above.