arendst/Tasmota #2216: sonoff basic ping loss / mqtt reconnect possible workaround

brnnrc, 2018-03-21:
Hi, 
I wrote this post as reply to #2178 but reading  other recent similar topics, I repost it as new issue. Sorry if it is wrong.
I'm facing a connectivity problem with a Sonoff basic (no problem with POW and S20 on equal configuration).
I have many ping lost with webserver and mqtt on (and nothing else) configuration. I get some MQT: Attempting connection... too.
RSSI 90%.
 It begins after the boot: I receive the first 5-6 ping reply and ping rate falls down to 60-70% packet loss.

When I boot without webserver, I have no ping loss or mqtt connection reconnection.

Disabling
//#define USE_ADC_VCC
or setting
#define APP_SLEEP 250

I have no ping loss even with webserver on!
I know it is very strange, but I tried multiple times ad I confirm.

I resoldered some joints but no change. I'm feeding 3.3v by vcc serial pin
I uploaded a simple arduino sketch with a webserver, mqtt and ADC_MODE(ADC_VCC); : no ping loss.

Hoping this helps to find the bug if any.

akasma74, 2018-03-21:
so there is no problem with web server, right?
you only changed some code, and it helped.
interesting.
had my SONOFF Basic going offline and back many times.
this morning replaced WiFiAP it uses with much better one, and had one or two offline cases.
was about to erase and re-flash the board..

the positive thing is the board seems to be fully functional after coming back online, and it usually takes it just a few moments.
I was thinking of modifying my notification in case of it going offline to wait for a minute, check if it's still offline and only then send a notification that the board is unavailable. still haven't implemented it though

brnnrc, 2018-03-23:
wonderful! I think other people should test this workaround too. Then I'd like to know why and if there is a bug (and a solution) :-)

akasma74, 2018-03-23:
To be honest, I don't quite get how increasing that value will decrease power consumption of the device (as stated in comments) - is it just because it will report its state less frequently?
Probably with default values it sometimes interferes with other operations and makes the whole device to glitch and drop connection? Just guessing.

On the side note, this morning I tried to press the button on my SONOFF while it was without its case. Had I light electric shock and an email that my device is unavailable. It rebooted itself and went back online quickly. I'm impressed ;)

brnnrc, 2018-03-23:
No, increasing that value permit the modem light sleep and so the power saving.
But I agree with you: the power saving itself can't explain better stability (I powered the 3.3v pin with a 2A power unit without success).

akasma74, 2018-03-25:
It restarted itself for the first time. Don't know how to dig deeper, but on the Information  page I can see:
`

Restart Reason | Exception
-- | --

`

Kepete, 2018-03-26:
Also experiencing Sonoff basic ping loss. Tried different firmware combinations. I disabled everything that wasn't needed, also MQTT. Only left webserver. I have two Unifi AP.

jjlawren, 2018-03-26:
I had issues with connectivity loss when running compiling against 2.4.0. After reverting to 2.3.0 it's been much more stable.

Kepete, 2018-03-27:
Tried to compile against 2.3.0 but didn't help. Webserver is not responding and ping loss about about 80%.

brnnrc, 2018-03-27:
@Kepete did you try to disable webserver? Did you try
Disabling
//#define USE_ADC_VCC
or setting
#define APP_SLEEP 250

Kepete, 2018-03-27:
@brnnrc Haven't tried to disable the webserver. 
Disabling #define USE_ADC_VCC didn't help and #define APP_SLEEP 250 seemed not to help also. 
I can switch the Sonoff manually with a switch connected to GPIO14 while the ping is not responding. So the Sonoff is not hanging, it just has some webserver or connectivity issues. Another Sonoff is sitting right next to the one with problems and is working correctly with most of the parameters defined.

brnnrc, 2018-03-27:
So @Kepete, it seems not like mine. Didi you try feeding by AC or by 3.3v pins? (I tried by 3.3v only)

Kepete, 2018-03-28:
@brnnrc  Yes tried now both ways. I can say it is better with Ac or 3.3v, both can have the problem for me.

brnnrc, 2018-03-28:
@Kepete ok, what is the rssi? It could be a wifi issue? I resoldered 2-3 critical points (eg. capacitors).

swap83, 2018-07-12:
I have same error. Sonoff Basic R2  and Tasmota 6.10. 

I try this workaround:
//#define USE_ADC_VCC --> Nothing

after  i set this:
#define APP_SLEEP 250 --> Working. 

The ping was always ok.

(sorry for my english)

wongnam, 2018-07-24:
My all sonoff basic, sonoff touch, bridge are facing this issue after i restart my wifi router.
all are using Tasmota version 6.1.1b, core 2.30, MQTT_ESPMQTTARDUINO, self compile by VSC.

symptom:
- Sonoff devices can reconnect to wifi router.
- Sonoff devices webserver can use normally
- Sonoff devices are no response to MQTT server on my Pi 3 B+. (no mqtt reconnection happen)
- restart Sonoff devices-> all will back to normal function as expect.