arendst/Tasmota #2217: Support APDS-9960 gesture/proximity/RGBW sensor

renne, 2018-03-21:
The [Avago APDS-9960](https://cdn.sparkfun.com/datasheets/Sensors/Proximity/apds9960.pdf) can be used for a lot of applications, e.g.

- gesture sensor for setpoint:

```
UP     dim
DOWN   dim
LEFT   colder
RIGHT  warmer
NEAR   on
FAR    off
```

- RGBW sensor for closed-loop light control of setpoint and real value (brightness and color temperature)

Adafruit offers [more information and an Arduino library](https://learn.adafruit.com/adafruit-apds9960-breakout/downloads).

renne, 2018-06-06:
What's the state of the [pull request](https://github.com/arendst/Sonoff-Tasmota/pull/566)?

ascillato, 2018-06-06:
Hi,

Sorry that your issue was not responded before. There were lots of issues.

I added the enhacement label to be taken into account.

Very nice chip! Thanks for share :+1:

renne, 2018-06-19:
I've added wiring information and an image in the discussion of #566

renne, 2018-07-01:
@Staars did you ever try to enable gesture, proximity and ambient light at the same time?

If we can get at least gestures and ambient light working at the same time I have an idea for a closed-loop light control:

1. set set-point for brightness (Lux) and color temperature (Kelvin) via MQTT in variables
2. [convert Kelvin to RGB](http://www.tannerhelland.com/4435/convert-temperature-rgb-algorithm-code/)
3. calculate upper and lower threshold of brightness and color temperature
4. write brightness thresholds into APDS-9960 registers (0x84-0x87)
4. enable ambient light and ambient light interrupt (register 0x80)
5. listen for ALS interrupts (register 0x93)

```
if (ALS interrupt) {
   unset ALS interrupt;
   if(brightness_current > brightness_setpoint) {
      Dimmer -= 1;
   } else {
      Dimmer += 1;
   }
   if(color_temperature_current > color_temperature_setpoint) {
      CT -= 1;
   }
   if(color_temperature_current < color_temperature_setpoint) {
      CT += 1;
   }
}
```

renne, 2018-07-01:
I don't want to use a hardware interrupt but read the interrupt status register to let the APDS-9960 detect when the brightness exceeds the thresholds. The closed-loop light control can be suspended as long as proximity/gesture is detected.

The idea is to set the brightness in Lux and the color temperature in Kelvin. Converting Kelvin to RGB is just for user convenience (slider instead of color picker, etc.). Tasmota would use the ambient light values of the APDS-9960 to adjust brightness and color temperature of the LEDs automagically. Node-Red is nice for testing. In real life light control should be independent from network outages.

Staars, 2018-07-01:
Okay, it is a bit clearer now. But no, I did not test it that way.
Maybe You can do some basic research here and write a sketch on a good old Arduino (as simple as possible) and test, if the sensor behaves that way. Then we can port that back to Tasmota.
Unfortunately I have little time at the moment . 

Regards 
Christian

sebadc, 2018-08-03:
Hi Guys. I am working whit an APDS-9960, I am try obtain the LUX and Color temperature with Adafruit library, but the result that I obtein is not real. If I read with a luxmeter the result is very diferent, Do you have any Examples with tested code?

My test code is:
#include "Adafruit_APDS9960.h"
Adafruit_APDS9960 apds;

void setup() {
  Serial.begin(115200);

  if(!apds.begin()){
    Serial.println("failed to initialize device! Please check your wiring.");
  }
  else Serial.println("Device initialized!");
  //enable color sensign mode
  apds.enableColor(true);
}

void loop() {
  //create some variables to store the color data in
  uint16_t r, g, b, c;
  //wait for color data to be ready
  while(!apds.colorDataReady()){
    delay(5);
  }
  //get the data and print the different channels
  apds.getColorData(&r, &g, &b, &c);
  Serial.print("red: ");
  Serial.print(r);
  Serial.print(" green: ");
  Serial.print(g);
  Serial.print(" blue: ");
  Serial.print(b);
  Serial.print(" clear: ");
  Serial.println(c);
  Serial.println();
  float luz=apds.calculateLux(r,g,b);
  float color=apds.calculateColorTemperature(r,g,b);
  Serial.print (" LUXES: ");
   Serial.println (luz);
  Serial.print (" Color: ");
   Serial.println (color);
   delay(500);
}
Regards. Seba

Staars, 2018-08-03:
This is really off topic now and you should ask at the Adafruit-Github-Repo.

Beside that I can not answer your question and as a general rule of a thumb, I do not expect professional grade performance from a 1-€-device which measures light-/color-data as a byproduct.
At least it is good to know, that we have strong evidence now to take these data with a grain of salt. So thank you for the report.

sebadc, 2018-08-03:
Thank for yours comments Christian

Staars, 2018-09-01:
@sebadc 

Hi, I revisited my code in the last days and for the first time, I really paid attention to light values.

My current „theory“ is, that the clear light value is (quite) correct, if you disable the gain (set AGAIN to 1x insteat of 4x).
Can you test it with your lux meter?

Best regards
Christian

renne, 2018-09-01:
@sebadc 

Can you try to read the raw sensor data with
```
uint16_t red    = 0;    /** RGB Red   value **/
uint16_t green  = 0;    /** RGB Green value **/
uint16_t blue   = 0;    /** RGB Blue  value **/
uint16_t lux    = 0;    /** Brightness in Lux unit **/

I2cValidRead16LE(&lux,   APDS9960_I2C_ADDR, APDS9960_CDATAL);
I2cValidRead16LE(&red,   APDS9960_I2C_ADDR, APDS9960_RDATAL);
I2cValidRead16LE(&green, APDS9960_I2C_ADDR, APDS9960_GDATAL);
I2cValidRead16LE(&blue,  APDS9960_I2C_ADDR, APDS9960_BDATAL);
```
and compare it with your luxmeter?

ascillato, 2018-09-06:
Hi,

APDS9960 is now supported by Tasmota by the PR: https://github.com/arendst/Sonoff-Tasmota/pull/3051

Just for reference: https://github.com/arendst/Sonoff-Tasmota/pull/566,

closed 2018-09-06 by ascillato2