arendst/Tasmota #2218: RF-Bridge-EFM8BB1 Firmeware

swampdogmash, 2018-03-21:
Hi Theo et al

I;ve been doing some stuff with the Sonoff Bridge and become very aware of how restricted it is with regard to the 433 Mhz protocols it supports. I am not sure if you have seen this project:

https://github.com/Portisch/RF-Bridge-EFM8BB1

Which is building a new firmware for the EFM8BB! chip that is on the sonoff bridge chip. It allows the chip to be flashed and for the 8266 firmware to ask it to receive other protocols than those implemented via the stock EFM firmware. this enables much greater flexibility and extends the usefulness of the Sonoff Bridge.

I thought it may be worth looking at adding support for the modified firmware into tasmota (additional serial commands are added to the EFM) to extend the supportable 433MHz devices with tasmota.

Sk4zz, 2018-03-21:
I support this request. I realize that there was a similar request about a month ago: [https://github.com/arendst/Sonoff-Tasmota/issues/1989](https://github.com/arendst/Sonoff-Tasmota/issues/1989) but the project has made significant progress since then.

swampdogmash, 2018-03-21:
In respect of the other issue, upgrading the firmware on the EFM was not as difficult as first thought. It's not necessarily a non-technical job, but many of the thing implemented in tasmota require external or technical capability to do them effectively. As the firmware seems to only require adjustment to the command sets used (and still supports the standard command sets that the original firmware supported) it didn't feel like a huge ask - but that will depend on memory required to implement the code (which could of course be optional given it is "special case") and could provide a solution to an issue that personally I have run into frequently - the lack of cheaper UK compatible light switches that run off only live and switched live connections (with no neutral). The alternatives are usually expensive (LightwaveRF, ZWave based) and thus kind of defeat the "DIY/cheap/maker" element of the 8266 movement.

Jason2866, 2018-03-21:
I am too, but Portisch Firmware is still in development
See #1989

akasma74, 2018-03-21:
swampdogmash, I was hoping to add some SONOFF Touch T1 to my setup, but it turned out I'm in the very same situation as yourself in terms of wires coming to light switches.
I also ended up ordering some VHome RF switches  and waiting for them to arrive.
However, apart from SONOFF RF I'm going to use RFLink flashed to Arduino Mega and commented to my RPi 3B. Apparently it's more capable than RF bridge in terms of flexibility and range of supported devices. And it's no more expensive than RF bridge as well.
My Arduino Mega arrived the country today, and I hope to play with it as soon as I receive it.
I will keep both RF gateways in use anyway because for reliability reasons (and of course because RF bridge is yet a device with limited set of supported protocols).