arendst/Tasmota #2219: Static ip configuration through the web interface

mizrachiran, 2018-03-21:
Hi Tasmota folks,

This is a feature request for adding option to set static IP using the web interface.
I would love to have such feature.

Cheers.

mizrachiran, 2018-03-22:
I'm aware of this command and used it before.
However, I think that for such a common configuration it would be much easier to be able to set it through the UI.

Jason2866, 2018-03-22:
I agree with @Frogmore42 fixed IPs are not needed by almost everyone.
This job is best handled by a a DHCP Server (in most cases does this a router)
And there is the possibility to assign always the same IP to a MAC adress.
If not trash the so called router in the bin

akasma74, 2018-03-22:
There is Static IP field in Espurna Web interface, and I filled it in to make ip address of my RF bridge static.
However, my Virgin Media Hub 3.0 assigned it a different address.. 
Don't know who to blame, but I just decided to stick with the hub's reservation rules to be on the safe side.

Jason2866, 2018-03-22:
@akasma74  This is Tasmota not Espurna!

akasma74, 2018-03-22:
@Jason2866 I know. My post is just an illustration of what @Frogmore42 and you explained to the TS.

knopserl, 2018-04-06:
You're all right to a ctain extent :-)
However, I like to have an overview of my home network and each device especially server or home devices have it's fixed static address only my mobile devives (phones and tablets) get dynamic addresses. 
I have even reserved ranges for certain classes (server, TV, PC, Home Automation) and from .0-.99 the DHCP devices are assigned.
It also does not allways work in every case thet you can connect via hostnam (but via IP-Adress always works) and static IP devices connect faster.
But in gernal, I have nothing against DHCP.

peteeee, 2018-06-09:
Yes here too I am interested in static IP configuration.  Today I do utilize DHCP reservation for a few devices but most of my network devices are using static IP.  (subnet mask here is full now at /25).
The Sonoff modes have introduced me to using Mosquitto and Node Red.  Current automation is using X10, UPB, ZWave and Zigbee.  I like what I can do with Mosquitton/Node Red.  

[image]

andrethomas, 2018-11-16:
@peteeee 

>  (subnet mask here is full now at /25)

You use live internet IP addresses for Tasmota devices? Just trying to understand why you have a /25 limitation?

andrethomas, 2018-11-16:
That aside I think the use of static ip addresses are a little deprecated and over complicates things for normal users. If you really want to use static IP addresses then there are commands that can be used for this as explained by @curzon01 and in the Wiki.

I do not think this particular feature would be in the best interest of 99% of Tasmota users.

@ascillato thoughts on this one?

ascillato, 2018-11-16:
Hi,

To use or not to use a fixed IP depends only on how you are configuring your network. You are free to use this feature or not. There are all the commands for setting that using the console, a mqtt message or a http command. 

But I think that the intention goes beyond that. It is on how easy is to set up Tasmota at first if you are using the standard precompiled bins. 

Now, first you flash your sonoff. 
Then you power it up and:

* If you send the wifi and mqtt broker config by serial, that is enough. The rest of configuration can be sent by mqtt and the ip doesn't matter.
* If you search the AP with your phone, you can set the wifi credentials. But then, as you can't configure the mqtt, you need to search the IP of your new sonoff in your router or using another program in order to finish the configuration.

So, for the second case, I think that should be useful to have more initial configuration fields in the AP mode like MQTT broker config and also fixed IP if you need it. At least all the parameters to config your Tasmota and not need to use your router or another program to continue the initial config.

andrethomas, 2018-11-16:
@ascillato 

Good point, will it be possible to give web console access in AP mode? That will solve a lot of other initial setup requirements.

chaosmaster, 2018-11-16:
@andrethomas I believe that already works, when going directly to http://<device-ip>/cs

peteeee, 2018-11-28:
**_You use live internet IP addresses for Tasmota devices? Just trying to understand why you have a /25 limitation?_**

No.

I still live in old world here of a small DHCP scope and many static IPs.  Home network is a bit over 100 devices today using multiple managed Gb switches talking to 4 networks from firewall.

andrethomas, 2018-11-28:
I use 192.168.32.0/20 - for static ip devices I just do static reservations on the DHCP server.

ShonP40, 2019-01-25:
I am trying to set 192.168.0.201 (my sonoffs current ip is 192.168.0.102) as my static ip on my sonoff but every time that i type: IPAddress1 192.168.0.201 it just says:
`15:15:49 CMD: IPAddress1 192.168.0.201
15:15:49 MQT: stat/ceilinglight/RESULT = {"IPAddress1":"192.168.0.201 (192.168.0.102)"}`

And its not switching the ip address
even after a restart.

ascillato, 2019-01-25:
Hi, seems that you have saving settings disabled. Please address this to the Tamota support chat. Thanks.

chat: https://discord.gg/Ks2Kzd4

ShonP40, 2019-01-25:
How can I enable saving settings?

ascillato, 2019-01-25:
Please, address this to the Tasmota Support Chat. It is more dynamic for questions and answers and for testings. Let's try not to go offtopic in this thread. Thanks.

crispy78, 2019-08-08:
I would also like this option. I’m a dedicated IP-guy. The DHCP-settings of my ISP provided momden/router can’t handle that many assigned IP-addresses and I can’t remember that many hostnames. Since a while the Fing app doesn’t provide that much info on the devices it finds. ipaddress1, ipaddress2, etc. does also work, but having this in the GUI would be a nice feature.

andrethomas2, 2019-10-13:
Closing this one for now since a simple workaround is posted here https://github.com/arendst/Sonoff-Tasmota/issues/2219#issuecomment-439521753

Can always be opened again later if there is demand for it.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-13 by andrethomas2

jeannotvh, 2019-12-12:
+1 for static IP as a nice-to-have. I would use that as well.
IMO, trusting the judgment of someone who asks for a feature is common courtesy. Especially when that feature is found in other products.

noormuhammedcjooma, 2020-01-23:
still not done?

Grikom, 2020-07-12:
to change ip in tasmota to 192.168.0.210

type in console:

ipaddress1 192.168.0.210
ipaddress2 192.168.0.1
ipaddress3 255.255.255.0

restart 1

amayii0, 2020-08-31:
First post when searching google for "Tasmota static ip"
I too need static IPs.
My ISP modem/router has a maximum DHCP range of 2-26. This fills pretty quick and I routinely assign IPs at 27 and above.

For the sake of it, [I'll copy/paste from doc](https://tasmota.github.io/docs/Commands/#wi-fi), it was quite obscure why ipaddress1/2/3/4 (yes, RTFM, but it usually starts with a Google search)

> IPAddress<x>	Set networking IP (XXX.XXX.XXX.XXX) addresses
> IPAddress1 = set device IP address
> 0.0.0.0 = use dynamic IP address (DHCP)
> XXX.XXX.XXX.XXX = set static IP address
> IPAddress2 = set gateway IP address
> IPAddress3 = set subnet mask
> IPAddress4 = set DNS server IP address
> 
> follow IPAddress commands with restart 1 to apply changes

mooizm, 2020-12-28:
Thanks for the info! It helped immensely on my coding. :)

muekno, 2021-01-24:
Its always the same, some like fixed IP, as me, some do not. DHCP is fine and the right thing for not so sophisticated users, More sophisticated users mostly prefer fixed IP. In my opinion servers and similar devices should have fixed IP, especially when you use an internal DNS server. Yes you can match mac addresses to  IP in most DHCP servers, so you have something like fixed IP, but in my now about more than 30 year in IT and networking I found out thats only about 95% stable, but for DNS and some other reasons, i.e. a server or a MQTT broker etc. it is essential the IP address is 100% fix and stable. DHCP ist ok for all the mobile Device and even for workstations and small home networks with an internet  router and  some workstations + some   mobile phones but the problem begins when there a network printer, then it is fine if the printers fixed IP ist known.
For the Sonoff device i.e. DHCP addresses are not a primary problem if the work as a MQTT client, as they can connect the broker from evrey IP, but the problem again begin if I will got to the Sonoff build in website and the IP hase changed, I have to go to my router or use NMAP to find the actuell IP and if I use DNS  I have to reconfigure it, not funny.
So I think to be able to set a fixed IP in Tasmota is a very meaningfull option.
Rainer

Jason2866, 2021-01-24:
@muekno  Tasmota supports fixed IP addresses. There is just no webui for this one time! needed setting.

manies2, 2021-02-26:
I've had a weird issue yesterday. I've been using Tasmota flashed Sonoff devices for about 18 months... (currently running 8.5.1) And my main router crashed yesterday. Meaning my DHCP server disappeared. 
I could still get to all my devices on my subnet with my HA (and MQTT) server, and switch on/off devices with HA. BUT for the life of me, I could not use ANY of the physical devices. 

It seems there is code trying to resolve the DHCP, before allowing the physical switch to operate...

I'm busy today switching ALL my devices to static IP's. And setting up DHCP on a sub-range, and static for the rest...

charredchar, 2021-03-06:
While I thank @amayii0 for directly posting the section of the manual to configure the IP via command line... I Highly disagree with the people posting stating giving accessibility options to others some how hinders the average user and I feel it is the complete opposite. WLED gives you the simple options to configure your network settings in the GUI but no one complains about that... If you have a tick option that enables advanced fields when disabling DHCP then it doesn't bother people who don't need it. I would consider WLED far more "complicated" than Tasmota but they get these no problem.

Just because you don't like static IPs on your network doesn't mean it is a bad idea to add the settings. This poor attitude drives people away, not makes them thank you for "keeping the settings simple." Because in the end needing to enter everything through a command line isn't simple.

jaymemaurice, 2021-03-06:
@charredchar I encourage you to fork, modify and build using gitpod. The process is well documented, free, and far easier than you probably expect. You can build any feature within your capability. If you are successful submit a pull request. Please keep in mind flash size is minimal and the main binary is getting big already. In your pull request, mention the minimal increase in size with quantitative numbers. While you are at it, make a box for NTP server since those who are not using DHCP will need it. I use DHCP with static leases to configure phones, IP cameras, thin clients, printers, network boot etc. because it’s far easier to configure everything on my router and replace the devices or change their network centrally.. I have no vested interest in this feature but I will offer support in the discussion thread and help review your pull request.