arendst/Tasmota #2220: Adding WebHook for status changes

mizrachiran, 2018-03-21:
Hi,

This is a feature request for adding a configurable WebHook http endpoint, to be invoked with the current status payload whenever something change.

Thanks

Jason2866, 2018-03-22:
Why?

mizrachiran, 2018-03-22:
When you control sonoff devices through http calls you don’t have the option to know if something change. 
The only alternative is polling which not as good.

Jason2866, 2018-03-22:
Ok i understand. The basic idea and concept from Tasmota is to control the devices via mqtt...
So if you want a solution via http Tasmota is the wrong choice.
I dont believe @arendst will invest much time and effort for doing it via http

mizrachiran, 2018-03-22:
Why?
It’s a great feature to have. 
I really think you should consider it.

mizrachiran, 2018-03-22:
There are lot of home automation hubs which doesn’t support mqtt. And some does not support polling either. 

Adding such feature would make Tasmota more powerful.

Jason2866, 2018-03-22:
No, it is not a great feature. Do you really read the arguments from @arendst in  #2222  ?

mizrachiran, 2018-03-22:
Yes, even before i opened this one. 
I still think it’s a great feature to have as it will enable many other scenarios for using Tasmota

mizrachiran, 2018-03-23:
@luisfpinto

luisfpinto, 2018-03-23:
Many current smart home and tech solution use Webhooks HTTP endpoint to send a message when a device status changed.  IFTTT, for example, uses this as a trigger for many devices. 

However, I can understand that Tasmota approach to this is more MQTT based but the solution pointed by @mizrachiran makes complete sense.

lobradov, 2018-03-23:
Why not cont

//_ on phone

lobradov, 2018-03-23:
(hate when this happens).

Why not contribute it yourself and maybe Theo will include it as optional
code?
if he already doesn't like the idea, he'll hardly do it on his own, but if
code is there...

FrankBolleri, 2021-01-01:
Hello All,
just a comment to help people searching for this matter and landing on this page in the future.
Seeing at #6692, invoke of an URL using rules currently is possible.