arendst/Tasmota #2223: Support for Homecube/Gosund SP1

knopserl, 2018-03-21:
It would be extremly attractive to get a support for the small good looking and cost effective smart plug SP1 sold from many different companies like Homecube, Gosund and others.
I have bought tow of them in a bundle for 39€ and the have an ESP8266EX with 1MB Flash and an Energy Monitor chip which is labeled with: HJL-01 J1749CYH D797480E.
I was able to solder the 3,3V and the RX/TX connection and read the chip type and flash size and dump the flash, but don't know what kind of energy monitor chip that is (have not found it in the web).
If anybody could hack that device and get Tasmota working, that would be great.
That are the links to the product:
[https://www.amazon.de/Steckdose-Homecube-intelligente-Verbrauchsanzeige-funktioniert/dp/B076Q2LKHG/ref=sr_1_fkmr0_1?ie=UTF8&qid=1521671695&sr=8-1-fkmr0&keywords=homecube+SP1 )](url)
or
[https://www.amazon.de/Intelligente-Stromverbrauch-Fernsteurung-Schaltbare-Energieklasse/dp/B076WZQS4S/ref=sr_1_1?s=diy&ie=UTF8&qid=1521671822&sr=1-1&keywords=gosund+SP1](url)

knopserl, 2018-04-03:
Meanwhile, with the help of another experience developer, we have found out the following inputs for the SP1:
This is the wiring of the used ESP8266-S3 board:
[image]
and the energy monitor chips is a HJL-01 which has the same wiring as the HLW8012 but need another driver code.

knopserl, 2018-05-13:
Is nobody interested in suppprting this very common large familiy of SP1 devices (from many different brands like: Gosund, Coosa, Homecube ...).
The only thing which is missing is the support of the energy IC: HJL-01

nightfly0815, 2018-06-25:
Hey it would be great if someone could add the device. There exists a lot of this devices with different names on Amazon for example. 
I ordered this.
https://www.amazon.de/Intelligente-Steckdose-Ablue-Funktioniert-Assistant/dp/B076DRFRZC/ref=sr_1_5?s=diy&ie=UTF8&qid=1529908498&sr=1-5&keywords=wlan+steckdose+smart+life
This device worked, not 100% with tasmota. It means that no configuration worked 100% for the device out of the box.
It was added in the newest espurna release named as blitzwolf-bwshp2

schrej-zz, 2018-06-26:
Espurna now supports these plugs including power monitoring.
https://github.com/xoseperez/espurna/issues/737
Would be great, if someone with coding knowledge could implement it in Tasmota.

arendst, 2018-06-26:
Did you try Tasmota selecting module 06 Sonoff Pow already?

arendst, 2018-06-26:
After troubling ordering the correct energy monitoring device with Amazon I'll see if Tasmota will work with it.

As far as I can tell this has NO energy monitoring:
https://www.amazon.de/Intelligente-Steckdose-Ablue-Funktioniert-Assistant/dp/B076DRFRZC/ref=sr_1_5?s=diy&ie=UTF8&qid=1529908498&sr=1-5&keywords=wlan+steckdose+smart+life

The below link has energy monitoring:
https://www.amazon.de/Steckdose-Homecube-intelligente-Verbrauchsanzeige-funktioniert/dp/B076Q2LKHG/ref=sr_1_fkmr0_1?s=diy&ie=UTF8&qid=1521672069&sr=1-1-fkmr0&keywords=homecube+SP1

I ordered the second one.

nightfly0815, 2018-06-26:
Yes I tried the pow module. I´m not 100 % sure how it worked. As I can remember I think the relay doesn´t worked when you switch on and off. The led also hasn´t worked with the pow configuration, and the energy monitor displays wrong values.
No you where wrong the Ablue has definitely the energy monitor inside!! I have checked it bevor with the original software. And the espurna blitzwolf.bin also worked well on the device with the energy monitor!!

Ok yes I think all the different brands with the same factor of the sockets have the same hardware inside. If it is mentioned in the description, then they have a energy monitor too.

knopserl, 2018-06-26:
I opened the SP1 thread in espurna and tasmota and the espurna team solved it first.
I guess the HLW8012 library with the seetings in hardware.h from espurna could be used for tasmota too, as the latest HLW8012 modification supports also the HJL01/BL0937 from the SP1 or similar devices. If someone does the work, tasmota could benefit and would support a plenty of other smart plugs with engery monitoring.

knopserl, 2018-06-27:
@arendst I would be surprised if the Ablue has no energy monitoring. I assume it's the same SP1 based product as used by: Gosund, Homecube, Coosa, Blitzwolf and many others companies (mostly the same product). If you open it and it has a HJL-01, it has energy monitoring

schrej-zz, 2018-06-28:
Yes, I think all of these low cost china plugs are utilizing this  HJL01/BL0937.
In Tasmota, you can use the on/off, but currently the values for power and voltage are crap.
This library might solve this as it did for Espruna.

gertmenke, 2018-06-28:
I was surprised when I ordered one with energy monitoring and found out that it was identical to two that I already had. Turns out, all three can do energy monitoring, although I am pretty sure that this feature was not advertised when I bought the first two. So, this was a pleasant surprise!

So I can confirm that these can do energy monitoring:
https://www.amazon.de/gp/product/B07CDCYLQ6/ref=oh_aui_detailpage_o00_s00?ie=UTF8&th=1
And these as well:
https://www.amazon.de/gp/product/B078LV94CZ/ref=oh_aui_detailpage_o01_s01?ie=UTF8&psc=1

gertmenke, 2018-06-28:
@arendst: The Ablue apparently has energy monitoring too, but it's quite well hidden further down in the description: "Senken Sie Ihre Rechnungen mit Energieüberwachung
Sie wissen genau, wie viel Strom jedes Gerät verwendet und wie effektiv Geräte verwaltet werden."

knopserl, 2018-06-29:
@arendst, @nightfly0815 as I wrote there are many (10+) companies selling it under different names but it is always a SP1.

arendst, 2018-07-01:
Working on implementation...

arendst, 2018-07-01:
How about this:

[image]

Thx to Xose for his ratio work.

Need some finishing touch before release.

gertmenke, 2018-07-01:
Awesome, thanks!

knopserl, 2018-07-01:
@arendst I'm impressed. You enabled a lot of device families with switch and energy monitoring. Probably not less than with support of the Sonoff devices.
Thanks a lot for adressing my proposal of this issue post. Seems that you made also a lot of other people happy!

closed 2018-07-02 by arendst

schrej-zz, 2018-07-03:
Try to use WPS. Worked for me.

nightfly0815, 2018-07-03:
I will try it, when I´m at home. But without any led notification it will be hard to find out how the device worked and in witch modus the device is. On witch GPIO is the standard led configured who shows the status of the device? Is it the GPIO 13? Maybe I can try to connect an extra led to the GPIO to indicate the status of the device.

schrej-zz, 2018-07-03:
Just enable WPS on your router and power up SP1. Should find an IP-Adress and you should find the sonoff-xxxx device in the listof devices.

nightfly0815, 2018-07-03:
Yes the WPS had worked!! It looks like that there is a wifi bug with tasmota and the socked.
Is there a way to set the red led too, when the socket is switched off?? (blue led on/ red led off)
The red led is orignal connected to the GPIO0.

knopserl, 2018-07-03:
@schrej-zz and @nightfly0815 WPS shall not be required as the SP1 will provide it's own hotspot when it can't connect to a another WiFi AP. It will offer a hotspot with 192.168.4.1. At least All 7 devices I have used (with espurna so far) behave that way and tasmota should have the same behavior.

nightfly0815, 2018-07-03:
@knopserl yes I know, all my other devices worked and open a wifi AP if the lost the normal wifi connection. This is why I asked that I don´t know this behavior that I can´t find the sonoff wifi from the socked.
I will test it on the second socked. Maybe there are some old firmware rests from the espurna firmware on the flash, because first I tried to flash the tasmota firmware over the espurna firmware in the webif of espurna. After that I erased the flash with esptool. For the second socket I will flash a blank 1mb file first with the espool. After this I will flash the new tasmota firmware. I report If the second socked worked without wps. What´s about the red led? Is it possible to use it?

arendst, 2018-07-03:
Tasmota default is WPS. Using the button as described in the wiki should make wifimanaged available. No visible feedback though.

myiotcenter, 2018-07-13:
Hi,
I flashed my SP1 smart socket with release 6.1.0 and having a problem with wifi connection...

`   Request timed out.
    Request timed out.
    Request timed out.
    Request timed out.
    Reply from 192.168.1.205: bytes=32 time=1ms TTL=128
    Reply from 192.168.1.205: bytes=32 time=2ms TTL=128
    Reply from 192.168.1.205: bytes=32 time=2ms TTL=128
    Reply from 192.168.1.205: bytes=32 time=1ms TTL=128
    Reply from 192.168.1.205: bytes=32 time=4ms TTL=128
    Reply from 192.168.1.205: bytes=32 time=15ms TTL=128
    Request timed out.
    Request timed out.
    Request timed out.
    Reply from 192.168.1.131: Destination host unreachable.
    Request timed out.
    Request timed out.
    Reply from 192.168.1.205: bytes=32 time=35ms TTL=128
    Request timed out.
    Reply from 192.168.1.205: bytes=32 time=1ms TTL=128
    Request timed out.
    Request timed out.
    Request timed out.
    Reply from 192.168.1.205: bytes=32 time=4ms TTL=128
    Request timed out.
    Request timed out.
    Request timed out.
    Request timed out.
    Request timed out.
    Request timed out.
    Reply from 192.168.1.205: bytes=32 time=135ms TTL=128
    Request timed out.
    Request timed out.
    Request timed out.
    Request timed out.
    Request timed out.
    Reply from 192.168.1.205: bytes=32 time=191ms TTL=128`

this socket is just few feet away  from my Ubiquity AP Pro so signal quality should not be an issue.

I used the source code and set my ip and ssid in user_config.h compiled in Arduino IDE and flash with esptool. Should I change the CFG_HOLDER value and recompile?  Anything specific I should look into ? 

Thanks !

schrej-zz, 2018-07-13:
I think you haven't searched for comparable issues.
Arduino IDE. Which version are you using.
Try flashing the precompiled version. Still issues?

crassus12, 2018-08-05:
Hello, is it possible under firmware 6.1.1 to integrate the HJL01 / BL0937 chip also under generic module? I can not find the chip in the selection.

Pfriemler, 2018-08-17:
@crassus12: As far as I understand, it's only possible to use "user gpios" - the GPIO_HLW_CF & GPIO_HLW_CF1 & GPIO_HLW_SEL are "ProgramSelectablePins" - not chooseable in the dropdown at all.
If you want to make more settings than the GPIO1/3 at the "BlitzwolF SHP2", try to change the "sonoff_template.h" and recompile.

Before I struggle with that myself, here are my questions:
a) as I understand, the red LED cannot be used as "switch on" led due to wiring issues?
b) the voltage measuring (probably all measurings) is stopped when the switch is off (-> 0V). In the original firmware, the plug continues to show voltage when off. What to I have to do get it back?
What does the original sonoff pow do when switch is off?

arendst, 2018-08-17:
b - look for the correct SetOption command in the wiki.

Pfriemler, 2018-08-17:
That was easy, BIG THANKS! SetOption21 was "OFF", but should be 1/on by default?
Nevertheless: it works. I have my mains reading in FHEM.

Mace404, 2018-09-15:
I flashed a Homecube SP1 with Tasmota but after flashing the led just blinks once when I plug it into a socket and nothing else happens.
Also used esptool to erase everything and flash again (and all those steps went fine) but same result.
Also verifying the flash with the binary gives a match.

> **esptool.py --port com4 verify_flash 0x0 sonoff.bin**
> esptool.py v2.5.0
> Serial port com4
> Connecting.....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Features: WiFi
> MAC: ec:fa:bc:93:f8:45
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Auto-detected Flash size: 1MB
> Verifying 0x75cd0 (482512) bytes @ 0x00000000 in flash against sonoff.bin...
> -- verify OK (digest matched)
> Hard resetting via RTS pin...

Any ideas?

mtdcr, 2018-09-15:
I guess the button doesn't work in Tasmota, so you can't switch between different configuration modes. I tried it with a Gosund SP1, which is probably identical to yours. Eventually I flashed espurna.

knopserl, 2018-09-15:
If you don't get it to work, I would also try ESPurna. we worked together to get all SP1 families running. It definitly works with Homecube, Gosund, Ablue, Coosa, Blitzwolf (I have tested it with all of them)

Pfriemler, 2018-09-16:
gosund SP1 uses a different GPIO (not the GPIO0 like sonoff devices), therefore you have to use a bridge on GPIO0 to GND to choose a wifi configuration mode, if you are unhappy with the default - which now is the WiFi-Manager beginning with 6.2.x.
After the gosund SP1 button is assigned (by using the BlitzWolf setting), the button can be used to set configuration mode if necessary - as usual with sonoff devices.

knopserl, 2018-09-16:
Hmmm, I had all devices open and they have all the same SP1 HW with the same ESP8266 board and same PCB,. I can't remember that different GPIO's for relay, button, LED's and energy monitor are used. But at least all work fine.

Mace404, 2018-09-17:
> gosund SP1 uses a different GPIO (not the GPIO0 like sonoff devices), therefore you have to use a bridge on GPIO0 to GND to choose a wifi configuration mode, if you are unhappy with the default - which now is the WiFi-Manager beginning with 6.2.x.

I added WiFi config the the user_config.sh script, so it should just start up and use those settings. 
Are you saying that when it's in a power socket I should short GPIO0 and GND instead of using the power button?

Mace404, 2018-09-18:
I didn't. But Pfriemler's comment suggests that to choose the config mode. 
As I edited the user_config.sh it should skip the WiFi manager but like I said, the device isn't doing anything but a short flash from the led after powering it.
Thanks for the instructions on how to configure it, but it won't come to that point even.
So there is no logging or whatsoever during startup? Hard to troubleshoot then. Guess I'll try espurna.

mtdcr, 2018-09-19:
Thanks for all the explanations! I had tried to use a version prior to 6.2. Nice to hear that current versions behave differently.

Pfriemler, 2018-09-22:
> So there is no logging or whatsoever during startup? Hard to troubleshoot then.

Of course there is. I used my CP2303 based USB adapter to flash and for configuration. Serial monitor is active for about 5 minutes.
Not sure, I think I used 115200 8N1. Or try 74888 and do a short power off (=reset). The ESP should show its boot messages, before switching to 115200.
I guess there is something gone wrong with the flash. Did you try that some times?

jult, 2018-10-08:
I currently have 4 GoSund Smart Socket SP1 units, the ones from https://www.amazon.de/gp/product/B07B911Y6V/ and just wondering if firmware flash is possible without the app contacting the cloud or breaking it open?
My units all say

"Firmware is the latest version
Wi-Fi module: 1.0.0
MCU module: 1.0.0"

and the app is Tuya Inc.'s Smart Life - Smart Living, which seems mostly open source.
The fact that they're still on firmware 1.0.0 seems odd to me. They're probably already vulnerable to remote access exploits and nobody at GoSund is doing anything about it.

schrej-zz, 2018-10-08:
@juilt: ...and what has this to do with Tasmota firmware?

schrej-zz, 2018-10-08:
No need to break them. You need a special tool to open the screws, that's it.
I replaced the screws afterwards.