arendst/Tasmota #2225: Sonoff basic /dual not always directly responces

NiekN, 2018-03-22:
I programmed a Basic and a Dual Sonoff device with Tasmota 
When i'm in the webintreface of the sonoff , you can toggle the relay very easy 
But i noticed that it is not always toggling directly. Mostly when i toggle it, it switched after 1 sec , but sometimes it  switches after 10sec or so , is that normal ? I also see this when im using the URL commandes in Chrome.  Is this a problem because the webinterface is also very slow ?

Im nearby the the router, so the wifi signal should be no problem.  

When using ewelink i never had this problem , it switches direcly

kpcz, 2018-03-22:
Have the same behavior, I think it is related to https://github.com/arendst/Sonoff-Tasmota/issues/1992

NiekN, 2018-03-22:
it not clear for me , it is solved by using a different MQTT_LIBRARY_TYPE ?

curzon01, 2018-03-23:
Could be a misunderstanding based on assumtions, so
- [ ] What hardware you are using
Sonoff Basic/Dual
- [ ] Provide the output of command ``status 0``

Did you flashed a binary from [releases](https://github.com/arendst/Sonoff-Tasmota/releases) or did you build your own from source?

NiekN, 2018-03-23:
Am using a Sonoff  Basic and a Dual (R2)

provide the output of command status 0 ?

i did not made a build of my own, i used this  
https://www.youtube.com/watch?v=pVPPiYAo8NI&t=1360s
esptool.exe -vv -cb 115200 -cp COM10 -ca 0x00000 -bz 1M -bm dout -cf sonoff.bin

curzon01, 2018-03-23:
Commands are explained in Wiki [Commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) and can be entered (besides the ither methodes described in Wiki) on webinterface under "Console".

On "Console" you can see what's logged
- try setting the log level for web log to more details using  the command `WebLog 3`
- stay on webinterface "Console" and observe what's going on

NiekN, 2018-03-23:
Im still using URL to control the sonoff like
http://myIP/cm?cmnd=POWER%20OFF
http://myIP/cm?cmnd=POWER OFF

im still not into the MQTT commands , i is still unclear for me  

You mean on the webpage of the sonoff, to check there the LOG?
But do i have to make a own build ?  or is the latest version of sonoff.bin the right one

curzon01, 2018-03-23:
You do not need your own build. You can use my suggestion commented above also with your build and yes, simply use "Console" button from webpage of your sonoff to take the command and observe the log.

NiekN, 2018-03-24:
18:38:59 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
18:39:11 MQT: Attempting connection...
18:39:24 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
18:39:35 MQT: Attempting connection...
18:39:49 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
18:40:01 MQT: Attempting connection...
18:40:15 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
18:40:17 RSL: domoticz/in = {"idx":12,"nvalue":1,"svalue":""}
18:40:17 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
18:40:17 RSL: stat/sonoff/POWER1 = ON
18:40:21 RSL: domoticz/in = {"idx":12,"nvalue":0,"svalue":""}
18:40:21 RSL: stat/sonoff/RESULT = {"POWER1":"OFF"}
18:40:21 RSL: stat/sonoff/POWER1 = OFF
18:40:25 RSL: domoticz/in = {"idx":13,"nvalue":1,"svalue":""}
18:40:25 RSL: stat/sonoff/RESULT = {"POWER2":"ON"}
18:40:25 RSL: stat/sonoff/POWER2 = ON
18:40:26 MQT: Attempting connection...
18:40:40 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
18:40:40 RSL: domoticz/in = {"idx":13,"nvalue":0,"svalue":""}
18:40:40 RSL: stat/sonoff/RESULT = {"POWER2":"OFF"}
18:40:40 RSL: stat/sonoff/POWER2 = OFF

NiekN, 2018-03-24:
looks like sonoff is busy with MQTT , which is not working

curzon01, 2018-03-25:
temporarly disable MQTT (e.g. using command `SetOption3 OFF`) - if this solves the problem you can enable MQTT again (`SetOption3 ON`) and try different MQTT_LIBRARY_TYPE 2 or 3

NiekN, 2018-03-25:
Hi curzon01

The sonoff is trying to connect to the MQTT broker, if he cannot find it , he retry in 10sec
Every retry gives a hiccup at the webpage/responce of relay.
I had found out that the  MQTT broker on the PI was not working . Also some settings in the sonoff where not correct. So i finally got it working   
 
Thx for help anyway

closed 2018-03-25 by NiekN