arendst/Tasmota #2226: Improvement - add TM1638 - 8 keys, 8 LEDs, 8 digital tubes

ionciubotaru, 2018-03-22:
Please consider adding support for [TM1638](https://www.banggood.com/TM1638-Chip-Key-Display-Module-8-Bits-Digital-LED-Tube-For-AVR-Arduino-p-928931.html?rmmds=search-left-hotproducts__2&cur_warehouse=CN) - 8 Bits, 8 keys, 8 LEDs, 8 digital tubes, common cathode LED digital tube. at only 2.43 EUR. The library is [here](https://github.com/rjbatista/tm1638-library).

migo75github, 2018-05-22:
Hi, this is a very good idea. When i can help you, let me know.  good day

ionciubotaru, 2018-05-22:
If your Arduino skills are good, try to add it to Tasmota, then ask @arendst to include it in main branch

arendst, 2018-06-28:
Ordered one. See if it's usefull...

ionciubotaru, 2018-06-28:
It's the cheapest 8 buttons + 8 leds + 8 digits interface on the market.
It's good to display temperature and to start different devices using the buttons - here is a potential application https://www.youtube.com/watch?v=ZFgh2KFLeGA&t=3s

arendst, 2018-06-28:
Just released experimental - untested switch support for the TM1638.

I haven't tested it yet as mine is still in the mail. What it should do is mimicking eight switches.

Pls give it a go and report any issues you may find.

ionciubotaru, 2018-06-29:
I tried this evening, unfortunately I didn't find any output when I press a switch nor in serial, nor in web console. 
`USE_TM1638` is enabled and DIO, STB and CLK pins are set on the web interface.
Module type = 18 - Generic

I will try again tomorrow .......

ionciubotaru, 2018-06-30:
This morning I tried again. 

First sketch:
```
#include <TM1638.h>
// define a module on data pin 8, clock pin 9 and strobe pin 7
TM1638 module(5, 4, 14);

void setup() {
  // display a hexadecimal number and set the left 4 dots
  module.setDisplayToHexNumber(0x1234ABCD, 0xF0);
}

void loop() {
  byte keys = module.getButtons();
  // light the first 4 red LEDs and the last 4 green LEDs as the buttons are pressed
  module.setLEDs(((keys & 0xF0) << 8) | (keys & 0xF));
}
```
works fine on esp8266 even at 3V3

next step, in TmLoop() ckecked if Tm1638GetButtons() works and it does, so, the problems are below this point.

```
void TmLoop()
{
  byte buttons = Tm1638GetButtons();
Serial.println(buttons);
  for (byte i = 0; i < 8; i++) {
    virtualswitch[i] = buttons &1;
    byte color = virtualswitch[i] ? TM1638_COLOR_RED : TM1638_COLOR_NONE;
    Tm1638SetLED(color, i);
    buttons >>= 1;
  }
  SwitchHandler(1);
}
```

Tm1638GetButtons() return:
0 - no button press
1 - button 1 is pressed
2 - button 2 is pressed
4 - button 3 is pressed
.......................

arendst, 2018-06-30:
Thx so far. Could you try to do a serial.print virtualswitch[i] too?

ionciubotaru, 2018-06-30:
modified sketch:
```
void TmLoop()
{
  byte buttons = Tm1638GetButtons();
Serial.print("buton ");
Serial.println(buttons);
  for (byte i = 0; i < 8; i++) {
    virtualswitch[i] = buttons &1;
Serial.print(i);
Serial.print(" - ");
Serial.println(virtualswitch[i]);
    byte color = virtualswitch[i] ? TM1638_COLOR_RED : TM1638_COLOR_NONE;
    Tm1638SetLED(color, i);
    buttons >>= 1;
  }
.........................
```
output: when button 2 pressed
```
buton 2
0 - 0
1 - 1
2 - 0
3 - 0
4 - 0
5 - 0
6 - 0
7 - 0
```

arendst, 2018-06-30:
Great!

Pls change line
```
    byte color = virtualswitch[i] ? TM1638_COLOR_RED : TM1638_COLOR_NONE;
```
in to
```
    byte color = (virtualswitch[i]) ? TM1638_COLOR_RED : TM1638_COLOR_NONE;
```
This should make a corresponding led light.

ionciubotaru, 2018-06-30:
I tried, it doesn't work. I think it's a problem in the sending chain:

```
void TmLoop()
{
  byte buttons = Tm1638GetButtons();
  for (byte i = 0; i < 8; i++) {
    virtualswitch[i] = buttons &1;
    byte color = (virtualswitch[i]) ? TM1638_COLOR_RED : TM1638_COLOR_NONE;
    Tm1638SetLED(color, i);
    buttons >>= 1;
  }
  SwitchHandler(1);
}
```

The following test doesn't work, while in example files works fine. 
```
void TmLoop()
{
  byte buttons = Tm1638GetButtons();
  // light the first 4 red LEDs and the last 4 green LEDs as the buttons are pressed
  Tm1638SetLEDs(((buttons & 0xF0) << 8) | (buttons & 0xF));
}
```

arendst, 2018-06-30:
Right. I suggest to hold on until I have mine so I can do some quick tests.

arendst, 2018-06-30:
Did include some delays as documented by https://github.com/MartyMacGyver/TM1638-demos-and-examples

Pls give it another try.

ionciubotaru, 2018-06-30:
Sorry it's not working. I checked again the library example and it works, even manually assign the pins to avoid any possible mistake and it doesn't change any led. Buttons read is ok.
Better wait for your device. Thank you for your work.

PS
Room thermostat is the first best choice
Display alerts will be another good option
Still search for a 3D printable enclosure to mount it on the wall.

Jason2866, 2018-07-13:
Received today :-)  
Will find some time this weekend to test....

ascillato2, 2018-08-23:
Implemented. Closing issue. Please, ask to reopen if needed. Thanks.

closed 2018-08-23 by ascillato2