arendst/Tasmota #2227: defining SYS_LOG_HOST causes an endless reboot loop

soif, 2018-03-22:
In v5.12.0h ( d89534164a7da98e54b19c864e38cb90191fe234 ) adding a hostname as `#define SYS_LOG_HOST "mysyslogserver.local"` in _user_config_override.h_ results in an endless reboot loop, when flashing (a previously erased) NodeMCU board. 

I can't swear but It seems that it worked fine in the v5.12.0c  ( 4c75873027c9f86859b5a4bbe83e440cb5918cf1 )

Here is the serial log:
```
00:00:00 Project sonoff nodemcu3 (Topic sonoff, Fallback DVES_149320, GroupTopic sonoffs) Version 5.12.0h-2_4_0
00:00:00 UPL: Arduino OTA Enabled Port 8266
00:00:00 WIF: Connecting to AP1 iot in mode 11N as sonoff-4896...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff-4896.local with IP address 10.1.209.243
14:41:52 MQT: Attempting connection...
14:41:53 MQT: Connected
14:41:53 MQT: tele/sonoff/LWT = Online (retained)
14:41:53 MQT: cmnd/sonoff/POWER = 
14:41:53 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0h","FallbackTopic":"DVES_149320","GroupTopic":"sonoffs"}
14:41:53 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-4896","IPAddress":"10.1.209.243"}
14:41:53 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
14:41:53 MQT: homeassistant/light/sonoff_1/config =  (retained)
14:41:53 MQT: homeassistant/switch/sonoff_1/config =  (retained)
14:41:54 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
14:41:54 MQT: stat/sonoff/POWER = OFF
14:42:01 MQT: tele/sonoff/STATE = {"Time":"2018-03-22T14:42:01","Uptime":"0T00:00:13","Vcc":2.937,"POWER":"OFF","Wifi":{"AP":1,"SSId":"iot","RSSI":82,"APMac":"66:70:02:B5:E1:D9"}}
14:42:11 WIF: Connect failed as AP cannot be reached
14:42:12 WIF: Connecting to AP1 iot in mode 11N as sonoff-4896...
14:42:18 WIF: Connect failed as AP cannot be reached
14:42:18 WIF: WPSConfig active for 3 minutes
14:42:22 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
```
it then reboots, and replay the same log.
during the startup the device is ping-able ONLY for 2 seconds (just after the AP1 has connected)

BTW if I don't define the syslog hostname at built time, but rather setup it later in the GUI, it woks fine.

curzon01, 2018-03-22:
Check if problem is remains if using IP instead domain name containing .local. 

.local domains can interfer with mDNS depending on your network environment.

see also [.local](https://en.wikipedia.org/wiki/.local) and [Multicast DNS Protocol overview](https://en.wikipedia.org/wiki/Multicast_DNS#Protocol_overview).

soif, 2018-03-23:
In fact i don't use **.local** names at all, but only FQDN hostnames driven by a local BIND based master DNS server, as well as 2 Slave BIND servers ... 😸  

And as stated b4, when defined from the GUI later, it works just fine.

But I will even try with IP only later : thanks for the tips.