arendst/Tasmota #2233: Adding Yeti to the Wiki repo

luisfpinto, 2018-03-23:
Hello! 
I've created a Yeti page on the wiki to show Tasmota users a new home automation app that supports Tasmota. Do I need to do any modification to add it to the wiki Sidebar?

Let me know if I need to add more information to the wiki page! 

Thank you!

reloxx13, 2018-03-24:
dunno who did it, but its added in the sidebar :)

bollitec, 2018-05-07:
@luisfpinto,

great working! 
Can you please append the information that the Tasmota device need for the scan weblog level 2 ?
Had a problem with one device and found that the log level was 0. The seems to use the web interface to get the data. Without web log level 2 the status information are not delivered.

luisfpinto, 2018-05-07:
Hey @bollitec Yeti expects the basic response: https://github.com/arendst/Sonoff-Tasmota/wiki/JSON-Status-Responses when you request the status information 

In case you have a user and password I haven't implemented this yet

closed 2018-06-04 by ascillato