arendst/Tasmota #2234: Sonoff Basic weird behaviour

LucaAghemo, 2018-03-23:
I'm using a Sonoff Basic with Tasmota 5.11.1 on a shade lamp, with a switch connected to GPIO14, Mosquitto and Home Assistant on RPI 3.
Flashing was flawless, everything worked for a few days, than it stopped. 
I could switch the lamp on and off with the manual switch (sometimes with a second of delay, sometimes immediate), the Sonoff was connected with the router and I could see the IP address, but I couldn't connect to the web page, the mqqt broker didn't connect and I couldn't control it with HA, that saw it as "Unavailable". With 4 presses the Sonoff went correctly in AP mode, I connected, I could only see the wifi connection page (that is already correct), I saved, it restarted and went back in this "coma" mode.
I'm using another Sonoff Basic with tasmota, that works flawlessly, so I thought it might be a hardware problem. Took a new Sonoff, flashed it, change the "coma" one with the new one, two or three days and... same behaviour!
Ok, i left it there for a while because I was busy in other stuff, after a few days it started working again! And then, a few days after, back to coma!
Meanwhile, the old Sonoff I unmounted and disconnected from everything remained unused for 10 days, I decided to reconnect it to see if something changed (without GPIO14 switch connected), but the behaviour is the same.
Honestly, I don't know what to do. I might flash another one but I'm afraid to fall in the same problem.
Any help?

disrupted, 2018-03-24:
+1 seeing the same issue with one of Basic's and Tasmota 5.12.0

WagnerBeccard, 2018-03-25:
Does it also click from time to time? Without any doing?

disrupted, 2018-03-25:
haha yes, exactly ! that's super weird and spooky

LucaAghemo, 2018-03-26:
yes, mine is also "clicking" sometimes, the Sonoff ghost... haha!
Ok, as now the only things we know is that 5.11.1 and 5.12 has the same problem.
Now it's a day and a half that it's working again, without doing anything but unplugging and leave it there for a while. BUT... when I plugged it back on, it didn't start to work immediatley, it took more or less two hours!
I'm worried my Sonoff will jump up and try to kill me during the night... loool!

WagnerBeccard, 2018-03-26:
I'm guessing the clicking is:
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#running-out-of-memory
That's why I start playing with platform, to disable some stuff. 
but now it feels like the switch is almost completely broken.. 
I get sometimes to the Webserver UI, but can't interact, 
or ERR_CONNECTION_TIMED_OUT
ll flash again without TLS, Hue bridge and wemo. maybe i can rescue it.

Barnabybear, 2018-03-26:
Hi, have a look at #2149

Kepete, 2018-03-26:
I have experienced lately something very similar to your problem. I had one Sonoff basic for a moth and it worked well. Now I got a second Sonoff basic and flashed it like I did the first one, downloaded Sonoff.bin and uploaded it using the ESPtool. I have a switch connected to GPIO14 on both Sonoffs.  Everything seemed to work fine at first but after a while the second Sonoff started hanging. 
The second Sonoff is slow and doesn't respond. It is hard to get access to the webserver interface. 
Both Sonoffs are running Tasamota 5.11.1 . Both have MQTT disabled and all other settings are the same.

WagnerBeccard, 2018-03-26:
took a look at #2149 
how do i commented out #define USE_DISCOVERY in platformIO?
Used // - not shure if this is the right way. would also like to know how to disable the wimo and hue component for saving memory. Maybe thats the issue.

Made the probably most boring video ever.
https://anonfile.com/SeQbU5d9bf/VID_20180326_202239.mp4
still the same issue. random restarts and no web UI. (without USE_DISCOVERY)
but with 4 presses the AP mode works well O_o

Kepete, 2018-03-26:
@WagnerBeccard  I did comment out it the same way. Sadly it didn't help for me either.

WagnerBeccard, 2018-03-26:
The sonoff ghost, its a memory problem.. 

The wiki already helps to fix this issue: 
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#esptool-usage

Best would be if we could clear region 0x0F4000 to 0x008000 via console..
             please?..

I hope i dont get this problem with the sonoff witch i put into the overhead ventilation

WagnerBeccard, 2018-03-27:
@Kepete 
Flashing the small FW allone didnt helped me as well.  You have to clear the memory first. 
My sonoff is back at the wall and doing his job again, now with a smaller FW. 

i did it with "esptool.py --port COM5 erase_region 0x0F4000 0x008000"

Kepete, 2018-03-27:
@WagnerBeccard  Thanks, I will take it out of the wall and try to erase the regions. Run a smaller FW on and I will update you.

LucaAghemo, 2018-03-27:
Thx a lot for the info. Actually for me it's a bit of a problem because:
1 - I need mqtt, as Home Assistant use it. Anything else (sensors, Domoticz etc.) is already commented, only exception the web server.
2 - I use a Mac and, for some reason, ESPtools doesn't work, so I don't know how to erase the memory. The weird thing is that, over 4 Sonoff that I have working with exactly the same firmware, only 2 behave this way, both of them mounted in the same lamp. Also, now it's about a week the Basic in that lamp is working flawlessly. A memory problem should repeteadly occur in similar situation, I can't understand what's going on here.
Anyway, I'll check deeply the link you proposed and see if I can sort out something.

WJ4IoT, 2018-04-12:
Hi, just my observations, I recently received a batch of true Sonoff Basic, why do I say true Sonoff basic? Not because they have a seal but the print says "Sonoff Basic R2 V1.0 2017.10.11 1750". This is where  my problems started; Wifi become unreliable in combination with MQTT (using Tasmota). My older two batches of sonoff basics are with a print "Sonoff TH V1.1 2017-05-05 J1712 or J1741"; these basics are still working flawlessly with Tasmota 5.12.0

Top layout except (print) and some holes soldered, looks identical, back is different (see pictures). 

My problem with the true Sonoff Basic:
17:16:12 MQT: Attempting connection...
17:16:12 MQT: Connected
17:16:12 MQT: tele/sonoff4/LWT = Online (retained)
17:16:12 MQT: cmnd/sonoff4/POWER = 
17:16:12 DNS: Initialized
17:16:23 WIF: Connecting to AP1 SomeWhereNearBy in mode 11N as sonoff4-3975...
17:16:27 WIF: Connected
17:16:27 DNS: Initialized
17:16:28 MQT: Attempting connection...
17:16:28 MQT: Connected

Reconnect to wifi after MQTT?
[image]
[image]

HobbytronicsPK, 2019-05-13:
Have a look on to [Ghost / Random Switching on Sonoff Devices (MQTT Retained)](https://blog.hobbytronics.pk/sonoff-tasmota-tips-tricks/#Ghost-Random-Switching)