arendst/Tasmota #2235: KMC 70011 Module can't quite fully calibrate

3ricj, 2018-03-23:
I've been trying to calibrate the power measurements on the KMC 70011 modules.. 

It seems the lowest value I can enter into PowerCal 4001,but it's still reading high by about 200 watts.  When I enter anything below 4001 it doesn't work: 

15:45:16 CMD: PowerCal 4000
15:45:16 MQT: stat/sonoff-2324/RESULT = {"PowerCal":12530}

3ricj, 2018-03-23:
Seems likely but it's hard to tell. Is this linear?

arendst, 2018-03-23:
If you compile yourself you might want to try and change the limit from 4000 to 3000 in file xdrv_03_energy.ino at line 881:
```
    if ((XdrvMailbox.payload > 0) && (XdrvMailbox.payload < 32001)) {
      Settings.energy_power_calibration = (XdrvMailbox.payload > 4000) ? XdrvMailbox.payload : HLW_PREF_PULSE;  // HLW = 12530, CSE = 5364
    }
```

3ricj, 2018-03-23:
3000 seems like ample

arendst, 2018-04-02:
Did you manage to solve your problem?

3ricj, 2018-04-02:
I've not been able to get a development environment setup so I can actually test this fix.  Is there a build somewhere I could try?

RrPt, 2018-04-07:
Hello,
i have the same problem with an board with identical components.
I calibrated with a 25W bulb

> 11:57:36 CMD: voltageset 230
> 11:57:36 MQT: stat/sonoff/RESULT = {"VoltageCal":119453}
> 11:58:12 CMD: powerset 25
> 11:58:12 MQT: stat/sonoff/RESULT = {"PowerCal":7948}
> 11:58:33 CMD: currentset 0.108
> 11:58:33 MQT: stat/sonoff/RESULT = {"CurrentCal":1500}
, and get this:
[image]
power and voltage look goot, but current is total wrong!
Then i switched to a 60W bulb and got this:
[image]
Power seems ok, but voltage ist too high!
Current is still wrong

Looks like the same Problem.
Can anybody help?
I can compile Versions wich changed code.
Regards Rainer