arendst/Tasmota #2236: Can somebody show the pinout for flashing of the new board of  1 Channel Switch 5V / 12V.

ronwar, 2018-03-23:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Jason2866, 2018-03-23:
??  shown in the pic ed1402 published on May 16 2017 ??

ronwar, 2018-03-23:
Listed on another issue (#453 ) 'New ESP8285 based 1Channel Inching/Locking 5V/12V Sonoff '

Jason2866, 2018-03-23:
In the issue #453 you mentioned is it described. Search for post from jktz90 commented on 7 Aug 2017

closed 2018-03-23 by arendst