arendst/Tasmota #2237: MH-Z19 Does not seem to connect

InvertedMinor, 2018-03-23:
[image]

Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

Jason2866, 2018-03-24:
Do you have disabled serial logging?

InvertedMinor, 2018-03-24:
yes

Jason2866, 2018-03-24:
Try to swap tx and rx. Sometimes it is false labeled.

Jason2866, 2018-03-24:
Do you have connected GND between sonoff and sensor?

tobox, 2018-03-26:
I had a MH-Z14 running with the MH-Z19 settings, and it was running nicely for quite a while. Then it sometimes started to have zero readings, with newer firmwares I always only received zero readings. 

I patched the firmware to use hardware serial (with serial.swap) instead of softserial, and now it is working flawlessly. But I would love to see this being fixed in a way so that the standard tasmota firmware continues working with MH-Z19 (and possibly MH-Z14a).

InvertedMinor, 2018-03-26:
Followed above comments: MH-Z19 connected to 5V power supply and measured the actual voltage, Gnd MH-Z19 connected to Sonoff, tried to switch rx/tx,  .... but no result (values remain at 0).......

mik13ST, 2022-08-19:
I can confirm you need hardware UART for MHZ14 to work as mentioned in https://github.com/arendst/Tasmota/issues/2237#issuecomment-376097913. You don't have to make you own tasmota build thought. If you pick the pins correctly, the MHZ driver will switch to hardware UART on its own, see this source code comment.
https://github.com/arendst/Tasmota/blob/c08561f67c722556655da7b8cbe1482d98aecd90/tasmota/tasmota_xsns_sensor/xsns_15_mhz19.ino#L26

This solution is already mentioned on the MHZ19 sensor page in the Wiring section (https://tasmota.github.io/docs/MH-Z19B/#wiring) so if people read the page carefully, unlike I did, resolving this should be straightforward.