arendst/Tasmota #2239: Sonoff 4CH PRO w/ BME280 & WS2812 Problem

GeorgeIoak, 2018-03-23:
I've detailed what I've tried to do over in the [Google groups](https://groups.google.com/forum/#!topic/sonoffusers/pfxjpMC3QnU) but no matter what combination I try the system will not boot if the WS2812 LED ring's Din pin is connected during boot. If I boot the system and then plug in Din everything works as expected.

I thought that it could have been a power issue but that doesn't appear to be the case based on my experiments. I tried moving the WS2812 to all 3 available GPIO pins (GPIO1,2,3) but none of them seem to work.

I have hardware serial disabled and if I use the web interface menu I can restart the system without a problem so the issue is only when power is first applied. I'm not sure if the system restart checks the boot pins or just restarts the code. When I say the system doesn't boot I mean that it does not get an IP address and the system starts with 1 or 2 relays active and does not respond to any button presses.

- [x] What hardware you are using? -  Sonoff 4CH PRO
- [x] Provide the output of command ``status 0``
```
10:48:41 CMD: status 0
10:48:41 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:48:41 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:00:53","Sleep":0,"BootCount":31,"SaveCount":118,"SaveAddress":"F8000"}}
10:48:41 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-18T14:23:28","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
10:48:41 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"TheIoaks","SSId2":"iotech-2.4GHz","TelePeriod":300,"SetOption":"000A0109"}}
10:48:41 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":474,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
10:48:41 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"GarageAFS","IPAddress":"192.168.10.231","Gateway":"192.168.10.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.10.1","Mac":"60:01:94:86:0A:7F","Webserver":2,"WifiConfig":3}}
10:48:41 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.10.201","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_860A7F","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:48:41 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Mar 23 18:48:41 2018","Local":"Fri Mar 23 10:48:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-8}}
10:48:41 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-03-23T10:48:41","BME280":{"Temperature":65.0,"Humidity":36.6,"Pressure":1020.3},"TempUnit":"F"}}
10:48:41 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-03-23T10:48:41","Uptime":"0T00:00:53","Vcc":3.464,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Dimmer":10,"Color":"25,25,25","Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"TheIoaks","RSSI":86,"APMac":"0E:02:8E:90:05:FC"}}}
```

Jason2866, 2018-03-24:
What signal levels uses the WS2812? ESP is 3.3 Volt device.
Things with TTL level (5 Volt)  may work or not. My advice use a level shifter.

GeorgeIoak, 2018-03-24:
The WS2812 are technically 5V devices but many people have shown that they can operate at 3.3V. In my situation I used the LED ring successfully at 3.3V and the issue is that the Sonoff does not boot if this LED ring Din pin is plugged in while booting.

Since WS2812 do not send data back to the "host" you only need to have the voltage level on Din meet the specification of the WS2812 (Vih = 0.7 * Vcc). There is no 2-way communication like in I2C and SPI so  there aren't any conditions where you could be damaging the ESP.

I updated my thread on Google Groups yesterday and found that an Adafruit NeoPixel strip will boot when plugged in. The die used in the LED Ring is obviously different as can be seen by the pictures I attached to the thread.

I am still trying to determine what could cause the Sonoff not to boot when this LED ring is plugged in yet if it is plugged in after the boot process it operates just fine.