arendst/Tasmota #2240: SonOff Basic Boot Fail

Paulf007, 2018-03-24:
I have a strange Problem with the sonoff basic.  
INFO :
- [ ] What hardware you are using. SONOFF Basic
- [ ] Provide the output of command ``status 0`` -  Not Booting 

I have a 2 Relay module connected to the sonoff and all works fine if the sonoff has booted up. But when you loose power to the sonoff it fails to boot unless you disconnect the relay boot the sonoff and then connect the relay module. 
Am I missing something? 
Best regards
Paul

Jason2866, 2018-03-24:
On which GPIO pins do you have connected the relays?
On boot up a few GPIO pins have to be defined high and low levels. If not the device wont start!

Paulf007, 2018-03-24:
Hi Jason

The relay is connected to 
GPIO1 and GPIO3 
Would that a difference?

Jason2866, 2018-03-24:
Yes, should work. Why not, I dont know.

curzon01, 2018-03-26:
Be aware GPIO1/3 are the serial line pins TX/RX - so also disable serial log