arendst/Tasmota #2241: Sonoff touch not connecting anymore

Sholofly, 2018-03-24:
After operating a few weeks succesfully, my sonoff touch is not working anymore. after upgrading to the latest tasmota firmware it worked foto a dat nut  now, the wifi led is blinking for a few seconds, stays on for a seconds, blinks twice and then turns off. This is repeating continuously. I can't connect tot any wireless interface anymore.. anyone an idea how to fix?

WagnerBeccard, 2018-03-26:
I think I had the same issue with my basic..
Try this out 

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#esptool-usage

Sholofly, 2018-03-27:
Thanks for the reply. I have switched it for a spare T1 version so can't test it for know. I'll try your suggestion later.

netruner, 2018-03-28:
I upgrade sonoff Touch from original system.
WiFi after upgrade signal strength is fatal ping 93% lost

Sholofly, 2018-04-26:
Touch is broken. Thank you @WagnerBeccard for you suggestion. No luck!

closed 2018-04-26 by Sholofly