arendst/Tasmota #2247: Power Strip?

Schnittcher, 2018-03-24:
I'm looking for a power strip that works with the Tasmota firmware.
Does somebody has any idea?

salopette, 2018-03-24:
Kann dir ein paar verkaufen!

Schnittcher, 2018-03-24:
Eigenbau?
Hast du mal ein Foto?

salopette, 2018-03-24:
https://www.ebay-kleinanzeigen.de/s-anzeige/wifi-wlan-smart-home-3-fach-steckdose-mit-usb-alexa-ios-android/832951655-176-2893

Schnittcher, 2018-03-24:
Kann man jede Steckdose einzeln ansteuern?

salopette, 2018-03-24:
Ja mann muss es als Generic Modul einrichten 
[URL=http://www.directupload.net/file/d/5036/x438wfw6_jpg.htm][IMG]http://fs1.directupload.net/images/180324/temp/x438wfw6.jpg[/IMG][/URL]

Schnittcher, 2018-03-24:
Cool, gibt es die auch mit mehr Steckdosen?

salopette, 2018-03-24:
Ist mir nicht bekannt

Schnittcher, 2018-03-24:
Also ist das eine, die man mit der Tasmota Firmware flashen kann und kein Eigenbau?

salopette, 2018-03-24:
Ja, kein Eigenbau und Flashbar!
Da esp verbaut

Schnittcher, 2018-03-24:
Kannst du mir eine bis morgen reservieren?

salopette, 2018-03-24:
Klar, kannst mir auch über Kleinanzeigen schreiben