arendst/Tasmota #2248: DS18B20 Calibration

chris86bbk, 2018-03-24:
Hi,

I've connected a DS18b20 to a S20 socket on gpio3.
Compared to other digital and analog sensors in the room the measured value mostly is two celsius too high.
How can I fix this?

Sensor is connected with 4,7k between VCC and DATA. I also tried parasitic mode, which did not solve the problem.

Jason2866, 2018-03-24:
The sensor is not accurate. You can do nothing...

erspearson, 2018-03-25:
The DS18B20 is actually pretty accurate - spec is +- 0.5C max error between -10C and +80C. I use them a lot and this seems true. It's a digital sensor so the connection method won't affect the readings; it'll either work or not work.
When you say "other ... in the room", are they in the exact same position? Like sitting right next to each other. If not the figures won't be comparable. Inside the Sonoff case will be hot as there's a PSU in there.

Of course, you might just have a bad one. You could try another. Maxim also sells certified ones that go through further testing but they're quite a lot more expensive.

curzon01, 2018-03-25:
Is the DS18b20 outside the S20 housing?

Frogmore42, 2018-03-27:
I agree, I have been using DS18B20 sensors for several years now and have found them to be very accurate.  I have also found that the temperature and humidity in a room varies significantly.  I can see the impact my furnace and fresh air intake (even when the furnace is not heating) have in a room that has no vents from the furnace.

If you have the DS18B20 close to your S20 it will be impacted by the 1W that the S20 is emitting as heat.