arendst/Tasmota #2250: MQTT in domoticz commands??

NiekN, 2018-03-24:
i have a sonoff dual and installed domoticz in a PI
With a URL command i can switch the dual on and off 

but i also what to check teh status . im try to use MQTT
i installed the MQTT broker, and it looks likes it is working ... 
when i toggle the dual in the Sonoff webpage, i see this in the logging at domoticz

2018-03-24 21:17:46.346 MQTT: Topic: domoticz/in, Message: {"idx":12,"nvalue":1,"svalue":""}
2018-03-24 21:19:22.474 MQTT: Topic: domoticz/in, Message: {"idx":12,"nvalue":0,"svalue":""}
2018-03-24 21:19:30.704 MQTT: Topic: domoticz/in, Message: {"idx":13,"nvalue":1,"svalue":""}
2018-03-24 21:19:34.018 MQTT: Topic: domoticz/in, Message: {"idx":12,"nvalue":1,"svalue":""}
2018-03-24 21:19:37.123 MQTT: Topic: domoticz/in, Message: {"idx":12,"nvalue":0,"svalue":""}
2018-03-24 21:19:37.894 MQTT: Topic: domoticz/in, Message: {"idx":13,"nvalue":0,"svalue":""}

when i toggle it at domoticz is see at the Sonoff webpage at console 
21:38:07 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
21:38:07 MQT: stat/sonoff/POWER1 = OFF
21:38:10 MQT: domoticz/in = {"idx":12,"nvalue":1,"svalue":""}
21:38:10 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
21:38:10 MQT: stat/sonoff/POWER1 = ON
21:38:12 MQT: domoticz/in = {"idx":13,"nvalue":1,"svalue":""}
21:38:12 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
21:38:12 MQT: stat/sonoff/POWER2 = ON
21:38:14 MQT: domoticz/in = {"idx":13,"nvalue":0,"svalue":""}
21:38:14 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
21:38:14 MQT: stat/sonoff/POWER2 = OFF
21:38:17 MQT: domoticz/in = {"idx":12,"nvalue":0,"svalue":""}
21:38:17 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
21:38:17 MQT: stat/sonoff/POWER1 = OFF

but how can i send MQTT commands in domoticz, and how can i get the status , so the switch will be updated?

arendst, 2018-03-25:
Why would you? It's already working from both the sonoff webpage and from domoticz as you noticed.

What else do you want?

NiekN, 2018-03-25:
my virtual switch  is not updating??

i checked again and i found somthing strange ......
when i toggled the sonoff at webpage indeed at domoticz the idx is also toggled but the wrong one
At domotixz Setup>Hardware the IDX is different at Setup>Devices  

[image]

[image]

 .

closed 2018-03-25 by arendst