arendst/Tasmota #2252: LED1 on Generic

vtochq, 2018-03-25:
Make sure these boxes are checked before submitting your issue - Thank you!

- [SmartLife 2 gang  switch] What hardware you are using
- [{"Status":{"Module":18,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":"0T00:00:59","Sleep":0,"BootCount":156,"SaveCount":1392,"SaveAddress":"F5000"},"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-02-09T14:54:51","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"},"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"dom2","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"5580000B"},"StatusMEM":{"ProgramSize":485,"Free":516,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3},"StatusNET":{"Hostname":"sonoff-4878","IPAddress":"192.168.11.102","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"2C:3A:E8:43:D3:0E","Webserver":2,"WifiConfig":1},"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_43D30E","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15},"StatusTIM":{"UTC":"Sun Mar 25 03:34:28 2018","Local":"Sun Mar 25 04:34:28 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1},"StatusSNS":{"Time":"2018-03-25T04:34:28"},"StatusSTS":{"Time":"2018-03-25T04:34:28","Uptime":"0T00:00:59","Vcc":3.149,"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"dom2","RSSI":54,"APMac":"D0:54:2D:10:A0:68"}}} ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

reloxx13, 2018-03-25:
i would just suggest to do a reboot if not done already.

vtochq, 2018-03-26:
Already reboot and also re-flash via serial.
If I map my LED1 to Relay1 output then my LED work properly (but without relay).
If I map LED and Relay to Relay1 output then works only one (i forgot relay or led). Can I use multi mapping one logical "output" to many GPIOs?

vtochq, 2018-03-26:
I find this function

void SetLedPower(uint8_t state)
{
  if (state) state = 1;
  digitalWrite(pin[GPIO_LED1], (bitRead(led_inverted, 0)) ? !state : state);
}

Is LED1 used not only for Switch1/Button1 indication?

KPK75, 2018-03-26:
Yes. Only LED1 is supported. Try setting leds as the same relay ( gpio2= relay2, gpio=relay3 ) and it should work.

KPK75, 2018-03-26:
Only LED1 is supported. So try setting leds as corresponding relays ( ie. gpio14 = relay3, gpio2 = relay2 )

manuelmouraguedes, 2019-02-13:
Also have this problem with SmartLife Switch 2/3Gang
Were you able to solve this? I would like to have leds linked to the relay status, red when on, blue when off like it works with the Tuya stock firmware
Is it possible to link leds to each of the relay status?

vtochq, 2019-02-14:
Tasmota is great thing, but if you want a lot customization it's not so good. Finally I used Espurna project to solve this problem.

manuelmouraguedes, 2019-02-14:
Cool, would you be so kind and share how you did it? Was a custom build? Can  you please share? Have the same 2/3 gang tuya wall switches and can’t make the leds work correctly according to the relays status  like they do with stock firmware
Many thanks!!!

vtochq, 2019-02-15:
Espurna builds a board-specific.

> Cool, would you be so kind and share how you did it? Was a custom build? Can you please share? Have the same 2/3 gang tuya wall switches and can’t make the leds work correctly according to the relays status like they do with stock firmware
> Many thanks!!!

Espurna builds for specific hardware.
This instruction for add support for this switch:
https://github.com/vtochq/custom/blob/master/Espurna_SmartLife_switch.md

manuelmouraguedes, 2019-02-15:
Thanks a lot!! Will try it

meingraham, 2019-02-15:
> I would like to have leds linked to the relay status, red when on, blue when off like it works with the Tuya stock firmware

Once you have the GPIO configuration assigned to the LEDs in the switch, you may be able to use a rule that follows the Power#State to turn on the corresponding LED.

However, if the LED's are controlled by a separate MCU from the ESP8266/ESP8285 specialized code which reverse engineers the original firmware will be required. It may be that control code for these LEDs has already been incorporated into TASMOTA. It may be a matter of selecting the appropriate BASE device. For finding out, I would recommend posting on the [TASMOTA Discord channel](https://discord.gg/Ks2Kzd4) or on the [Sonoff Users Google Group](https://groups.google.com/forum/#!forum/sonoffusers)? 

Otherwise, a code tweak to one of the TASMOTA driver modules or code in ESPurna, special code for the device will be required. Perhaps ESPurna has already added support for this device.

Mike

melikyuksel, 2019-03-15:
@meingraham Is there even a command to control LEDs other than Led1? There doesn't seem to be.

meingraham, 2019-03-15:
What you do is assign a RelayX to the GPIO to which the LED is connected to. Then you use PowerX command associated with the "relay" to control the LED.