arendst/Tasmota #2253: Constantly disconnect and connect from MQTT broker

dzinyit, 2018-03-25:
I'm using mosquitto as mqtt broker on hass.io. i refered to troubleshoot and it suggested there be two devices with same client ID, however, I'm pretty sure there is only one sonoff connected and the client id is unique, yet the two clients are with different "keepalive"s, respectively 15 and 60. any help is appreciated!

configuration.yaml:
```
mqtt:
  broker: 192.168.1.154
  port: 1883
  client_id: Sonoff_modified_on_3_20_2018
  keepalive: 60
  username: sonoff
  password: sonoffpwd
  protocol: 3.1
  birth_message:
    topic: "tele/hass/LWT"
    payload: "Online"
    qos: 1
    retain: true
  will_message:
    topic: "tele/hass/LWT"
    payload: "Offline"
    qos: 1
    retain: true
light:
  - platform: mqtt
    name: "Sonoff Switch 01"
    state_topic: "stat/sonoff/POWER"
    command_topic: "cmnd/sonoff/POWER"
    availability_topic: "tele/sonoff/LWT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    payload_available: "Online"
    payload_not_available: "Offline"
    retain: true

```
mqtt log:

> starting version 3.2.2
> 1521956842: mosquitto version 1.4.12 (build date 2017-06-01 13:03:48+0000) starting
> 1521956842: Config loaded from /etc/mosquitto.conf.
> 1521956842: Opening ipv4 listen socket on port 1883.
> 1521956842: Opening ipv6 listen socket on port 1883.
> 1521956842: Warning: Mosquitto should not be run as root/administrator.
> 1521956858: New connection from 192.168.1.113 on port 1883.
> 1521956858: New client connected from 192.168.1.113 as Sonoff_modified_on_3_20_2018 (c1, k15, u'sonoff').
> 1521956870: New connection from 192.168.1.113 on port 1883.
> 1521956870: Client Sonoff_modified_on_3_20_2018 already connected, closing old connection.
> 1521956870: Client Sonoff_modified_on_3_20_2018 disconnected.
> 1521956870: New client connected from 192.168.1.113 as Sonoff_modified_on_3_20_2018 (c1, k15, u'sonoff').
> 1521956876: New connection from 192.168.1.154 on port 1883.
> 1521956877: Socket error on client <unknown>, disconnecting.
> 1521956877: New connection from 192.168.1.154 on port 1883.
> 1521956877: Client Sonoff_modified_on_3_20_2018 already connected, closing old connection.
> 1521956877: Client Sonoff_modified_on_3_20_2018 disconnected.
> 1521956877: New client connected from 192.168.1.154 as Sonoff_modified_on_3_20_2018 (c1, k60, u'sonoff').
> 1521956878: New connection from 192.168.1.113 on port 1883.
> 1521956878: Client Sonoff_modified_on_3_20_2018 already connected, closing old connection.
> 1521956878: Client Sonoff_modified_on_3_20_2018 disconnected.
> 1521956878: New client connected from 192.168.1.113 as Sonoff_modified_on_3_20_2018 (c1, k15, u'sonoff').
> 1521956878: New connection from 192.168.1.154 on port 1883.
> 1521956879: Socket error on client <unknown>, disconnecting.
> 

sonoff log:

> 06:48:04 MQT: stat/sonoff/POWER = ON
> 06:48:06 MQT: Attempting connection...
> 06:48:06 MQT: Connected
> 06:48:06 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:06 MQT: cmnd/sonoff/POWER = 
> 06:48:08 MQT: Attempting connection...
> 06:48:08 MQT: Connected
> 06:48:08 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:08 MQT: cmnd/sonoff/POWER = 
> 06:48:08 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
> 06:48:08 MQT: stat/sonoff/POWER = ON
> 06:48:10 MQT: Attempting connection...
> 06:48:10 MQT: Connected
> 06:48:10 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:10 MQT: cmnd/sonoff/POWER = 
> 06:48:10 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
> 06:48:10 MQT: stat/sonoff/POWER = ON
> 06:48:12 MQT: Attempting connection...
> 06:48:12 MQT: Connected
> 06:48:12 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:12 MQT: cmnd/sonoff/POWER = 
> 06:48:12 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
> 06:48:12 MQT: stat/sonoff/POWER = ON
> 06:48:14 MQT: Attempting connection...
> 06:48:14 MQT: Connected
> 06:48:14 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:14 MQT: cmnd/sonoff/POWER = 
> 06:48:14 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
> 06:48:14 MQT: stat/sonoff/POWER = ON
> 06:48:16 MQT: Attempting connection...
> 06:48:16 MQT: Connected
> 06:48:16 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:16 MQT: cmnd/sonoff/POWER = 
> 06:48:16 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
> 06:48:16 MQT: stat/sonoff/POWER = ON
> 06:48:18 MQT: Attempting connection...
> 06:48:18 MQT: Connected
> 06:48:18 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:18 MQT: cmnd/sonoff/POWER = 
> 06:48:18 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
> 06:48:18 MQT: stat/sonoff/POWER = ON
> 06:48:20 MQT: Attempting connection...
> 06:48:20 MQT: Connected
> 06:48:20 MQT: tele/sonoff/LWT = Online (retained)
> 06:48:20 MQT: cmnd/sonoff/POWER = 
> 06:48:20 MQT: stat/sonoff/RESULT = {"POWER":"ON"}
> 06:48:20 MQT: stat/sonoff/POWER = ON

sonoff info:

> Program Version | 5.12.0
> -- | --
> 2018-03-20T15:50:39
> 2_4_0/2.1.0(deb1901)
> 0T00:02:26
> 114 at FA000
> 55
> Power on
> Sonoff
>  
> DzinYit (100%)
> sonoff-5633
> 192.168.1.113
> 192.168.1.1
> 255.255.255.0
> 192.168.1.1
> 5C:CF:7F:72:96:01
>  
> 192.168.1.154
> 1883
> Sonoff_modified_on_3_20_2018
> sonoff
> sonoff
> sonoffs
> cmnd/sonoff/
>  
> None
> Enabled
> Web Server
>  
> 7509505
> 1327198
> 1024kB
> 1024kB
> 495kB
> 508kB
> 20kB
>

lobradov, 2018-03-25:
> 1521956877: Socket error on client , disconnecting.
> 1521956879: Socket error on client , disconnecting.

Your Sonoff and your Home Assistant are connecting using same client id.

dzinyit, 2018-03-25:
> Your Sonoff and your Home Assistant are connecting using same client id.

oh, stupid me, i copied the sonoff name to the HA, many thanks!!

closed 2018-03-25 by dzinyit