arendst/Tasmota #2254: Hi,

AlexTakt, 2018-03-25:
Make sure these boxes are checked before submitting your issue - Thank you!

- [ ] What hardware you are using
- [ ] Provide the output of command ``status 0``
- [ ] If you have a stack dump decode it: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst
- [ ] For better debug messages: https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

AlexTakt, 2018-03-25:
Hi,
Thank yo for this product.
I'm using.
Sonoff Basic Module
Program Version | 5.12.0

Svitching is working well/ However whel I added DS18B20, i found the issue.
Readings are OK and web interface is updating as expected, but MQTT seems not publishing any temperature updates. No by temperature changed, nor by any reasonable  period. I didn't find any config on that, is there command to start getting MQTT updates on Temperature change?

Thanks.

PS
09:15:44 CMD: status 0
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff-01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","Uptime":"0T00:40:46","Sleep":0,"BootCount":46,"SaveCount":959,"SaveAddress":"F5000"}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-03-03T21:56:27","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId1":"DT","SSId2":"NAS","TelePeriod":300,"SetOption":"00000009"}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS4 = {"StatusMEM":{"ProgramSize":494,"Free":508,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-01-4697","IPAddress":"192.168.21.60","Gateway":"192.168.21.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.21.1","Mac":"2C:3A:E8:45:52:59","Webserver":2,"WifiConfig":3}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.21.22","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_455259","MqttUser":"alex","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 25 08:15:44 2018","Local":"Sun Mar 25 09:15:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS10 = {"StatusSNS":{"Time":"2018-03-25T09:15:44","DS18B20":{"Temperature":22.2},"TempUnit":"C"}}
09:15:44 MQT: tasmota/sonoff-01/stat/STATUS11 = {"StatusSTS":{"Time":"2018-03-25T09:15:44","Uptime":"0T00:40:46","Vcc":3.400,"POWER":"0","Wifi":{"AP":1,"SSId":"DT","RSSI":80,"APMac":"00:50:7F:79:F3:B8"}}}

AlexTakt, 2018-03-25:
Looks like I'm not quite correct.
There are MQTT temperature updates every 5 min. 
Ith there option to change this time, or better yet send MQTT updates on temperature change?
Thanks.

curzon01, 2018-03-25:
TelePeriod

AlexTakt, 2018-03-25:
Thank you!
By any chance, do you know the way to get those updates on temperature change event?

curzon01, 2018-03-25:
You can get the periodical change by TelePeriod but not an event by change of temp value.

AlexTakt, 2018-03-26:
Thank you!
This part works as you told.

closed 2018-03-26 by AlexTakt