arendst/Tasmota #2256: WIKI should reflect the change in naming of LwIPVariant in 2.4.1

TRIROG, 2018-03-25:
According to the wiki Arduino IDE tutorial, in chapter Configure Arduino IDE it is stated that one should select v1.4 Prebuilt in the board manager menu, however in the 2.4.1. version this is called v1.4 Higher Bandwidth which can be confusing.

moose4lord, 2018-03-30:
That confused me too. I  originally selected "v1.4 compile from source" and got the error "exec: 'make': executable file not found in %PATH%". Switching to "v1.4 Higher Bandwidth" fixed it.

kolbenhans, 2018-04-03:
i can confirm that. unlike described in the documentation (https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE), the selection for "v1.4 Prebuild" no longer possible, instead there are the entries:" v1.4 compile from source" & "v1.4 Higher Bandwidth"

if you select entry: v1.4 compile from source you get the error "exec: 'make': executable file not found in %PATH%". If you select entry "v1.4 Higher Bandwidth" you no longer get this error, but instead: 
exit status 1
Error compiling for the board generic ESP8266 module