arendst/Tasmota #2257: sonoff pow day time from 25th mar 7:am to 26th mar 7:am -  help wanted

earth08, 2018-03-25:
Dear, I would like to know,
can i set my sonpow day saving like 
yesterday: 24th march 7am to 25th march 7am
today: 25th march 7am to 26th march 7am 

Is there any any way i can get tis results,
Please guide

Make sure these boxes are checked before submitting your issue - Thank you!

- [ * ] What hardware you are using
SonOff Pow
- [ ] Provide the output of command ``status 0``
15:11:24 CMD: status 0
15:11:24 MQT: stat/home/first/pinu/pow/STATUS = {"Status":{"Module":6,"FriendlyName":"Pin - PowerSwitch 101","Topic":"home/first/pinu/pow","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:42","StartupUTC":"2018-03-25T13:06:42","Sleep":0,"BootCount":26,"SaveCount":97,"SaveAddress":"FB000"}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS2 = {"StatusFWR":{"Version":"5.12.0f","BuildDateTime":"2018-03-22T18:44:21","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"-------------","LogPort":-----,"SSId1":"Jay ","SSId2":"Jay 1 ","TelePeriod":300,"SetOption":"00020009"}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS4 = {"StatusMEM":{"ProgramSize":496,"Free":504,"Heap":19,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS5 = {"StatusNET":{"Hostname":"home/first/pinu/pow-7360","IPAddress":"------------","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.11.1","Mac":"----------","Webserver":2,"WifiConfig":3}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.11.33","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_071CC0","MqttUser":"kiran","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS7 = {"StatusTIM":{"UTC":"Sun Mar 25 13:11:24 2018","Local":"Sun Mar 25 15:11:24 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS10 = {"StatusSNS":{"Time":"2018-03-25T15:11:24","ENERGY":{"Total":2.370,"Yesterday":1.653,"Today":0.016,"Power":2,"Factor":0.04,"Voltage":246,"Current":0.189}}}
15:11:24 MQT: stat/home/first/pinu/pow/STATUS11 = {"StatusSTS":{"Time":"2018-03-25T15:11:24","Uptime":"0T00:04:42","Vcc":3.472,"POWER":"ON","Wifi":{"AP":2,"SSId":"Jay Ambe 1","RSSI":48,"APMac":"34:CE:00:59:AC:99"}}}

Phrede, 2018-03-25:
The daily energy is calculated from midnight (00:00).   If you need values based on 7AM instead why not just change the timezone setting by +7?  The time on the report may be inaccurate but the data would be correct and the time could be corrected.

earth08, 2018-03-26:
Please check following
time resets at around 3:20 i have set my time for india by adjusting daytime while compiling and setting timezone to 99 as suggested by arendst 

[image]
[image]