arendst/Tasmota #2259: Sonoff Pow - Strange readings

jelee1988, 2018-03-26:
- [ ] What hardware you are using Sonoff
 POW
- [ ] Provide the output of command ``status 0``
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS = {"Status":{"Module":6,"FriendlyName":"Sonoff","Topic":"pow1/keuken/koelkast","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","Uptime":"0T06:38:04","StartupUTC":"","Sleep":0,"BootCount":254,"SaveCount":504,"SaveAddress":"F4000"}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS2 = {"StatusFWR":{"Version":"5.12.0e","BuildDateTime":"2018-03-13T00:03:38","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId1":"test123","SSId2":"","TelePeriod":120,"SetOption":"000A0009"}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS4 = {"StatusMEM":{"ProgramSize":520,"Free":480,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS5 = {"StatusNET":{"Hostname":"pow1/keuken/koelkast-3956","IPAddress":"192.168.1.22","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"EC:FA:BC:00:AF:74","Webserver":2,"WifiConfig":3}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.13","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_00AF74","MqttUser":"homeassistant","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 09:11:43 1970","Local":"Thu Jan 01 09:11:43 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS9 = {"StatusPTH":{"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T09:11:43","ENERGY":{"Total":11.579,"Yesterday":2.042,"Today":9.139,"Power":0,"Factor":0.00,"Voltage":219,"Current":0.000}}}
09:11:43 MQT: stat/pow1/keuken/koelkast/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T09:11:43","Uptime":"0T06:38:04","Vcc":3.466,"POWER":"ON","Wifi":{"AP":1,"SSId":"test123","RSSI":92,"APMac":"10:7B:EF:B9:F5:11"}}}
09:11:52 MQT: tele/pow1/keuken/koelkast/STATE = {"Time":"1970-01-01T09:11:52","Uptime":"0T06:38:13","Vcc":3.469,"POWER":"ON","Wifi":{"AP":1,"SSId":"test123","RSSI":86,"APMac":"10:7B:EF:B9:F5:11"}}

- [ ] Software
Program Version | 5.12.0e

Got a Sonoff POW on me fridge, but getting strange value's (KwH yesteryday) 
got is 7 days running now, this are the value's;

Voltage 0.000 A
Current 0 W
Power 0.00
Energy Today 9.139 kWh 
Energy Yesterday 2.042 kWh ( full 24 hours lower then energy_today?)
Energy Total 11.579 kWh

curzon01, 2018-03-26:
Yesterday is not a moving 24h window, it is yesterday resetting at midnight. So your fridge consumed more power today (since midnight) than yesterday (before midnight), that's might be ok.

If values are incorrect, did you [calibrate](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Pow#calibration) your Pow?

jelee1988, 2018-03-27:
Today 10.444 kWh
Yesteryday 2.042 kWh
Total 12.884 kWh

Stil strange value's, 

The Energy Yesteryday, does not change in value. 

I have calibrate the Pow

curzon01, 2018-03-27:
Why should yesterday change, yesterday is gone. And the value, so it's a large fridge...

jelee1988, 2018-03-27:
Readings 25/03/2018 
Energy Today 9.139 kWh
Energy Yesterday  2.042 kWh

Readings 26/03/2018
Energy Today: 10,026 kWh
Energy Yesterday : 2.042 kWh

Readings Today:  
Energy  Today 10.444 kWh
Energy  Yesteryday 2.042 kWh

I was suspecting something like this: 
Energy  Today 10.444 kWh
Energy  Yesterday : 10,026 kWh (Value of "energy_Today" 26/03/2018)  
Energy_Total (combined ammount ALL TIME) 

it's a bit confusing for me right now

Frogmore42, 2018-03-27:
Your POW does not have the right time:
"Time":"1970-01-01T09:11:52"
There are a few reasons why that can be.  There are several issues with things to try and resolve the issue.  Assuming that you have the NTP servers correctly configured and that the device can reach the Internet to talk to them, I have found that unplugging the device for a half hour or more (remove ALL power from the device, so it is completely off and the RTC can reset completely) and then plugging it in was enough to get it to connect.  Before I discovered that trick the only thing that worked was reflashing the device.