arendst/Tasmota #2260: Using basic ESP8266 to react to reed switch

mattpurnell, 2018-03-26:
Apologes if this topic is answered many times already; every search I do with 'esp8266' obvioulsy returns many false positives.

I love the Tasmota firmware; so much so I have applied it to a basic ESP8266 unit like this one below (image), in order to keep all my switches and sensors standard

Everything is fine EXCEPT I cannot seem to assign GPIO0 or GPOIO2 - which are the only two pins exposed (other than the Power/Reset/TX/RX).

Ultimately I want the MQTT message sent if a Pin changes from LOW to HIGH; this will be used for my Home Assistant to know (for example) if the front door has been opened - via REED switch.

Is this possible? Thanks again to this awesome community!

[image]

curzon01, 2018-03-26:
You can't quit use GPIO0/2 because these pins are controlling the bootloader mode:

  | GPIO 0 | GPIO 2 | GPIO 15
-- | -- | -- | --
UART Download Mode (Programming) | 0 | 1 | 0
Boot from Flash (Normal startup) | 1 | 1 | 0
Boot from SD-Card | 0 | 0 | 1

see also [ESP 8266 Boot Process](https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process)

But your assumtion not using TX/RX is wrong: You can use TX (GPIO1) and RX (GPIO3) if you disable serial logging - I always did it in the past using ESP-01 (your device) with Tasmota

mattpurnell, 2018-03-26:
@curzon01 Awesome, thanks for that tip!

closed 2018-03-26 by mattpurnell