arendst/Tasmota #2261: 2 Pressure Sensors (Differential Pressure) with I2C (BMP180 and BMP280)

Bastian86, 2018-03-26:
Hello I am using a Sonoff Basic and would like to use two pressure sensors BMP180 and BMP280 as differential pressure over MQTT and FHEm.
Separately, the sensors can communicate with the Sonoff Basic. If I connect both, the Sonoff sees only the BMP180. I use the current original version 5.12.0 d
Is it possible to expect support for these two sensors at the same time in the future?

I have read:
Two SHT3X on I2C bus - feasible? #1949
I2C support multiple of the same sensor type #1049

martinlindjarv, 2018-03-26:
Hello, Just guessing but can it be problem that both sensors use same I2C address?
And different sensors need different addresses on same bus?

Bastian86, 2018-03-27:
Hello, the I2C addresses are different. That's why I chose two different types.
What does not work are two sensors of the same type so BMP180 or BMP280, since they use the same address.
BMP180 0x77h
BMP280 0x76h
If I connect both sensors, I only see BMP180.

Is it possible to expect support for two sensors at the same time in the future?

johngouk, 2018-05-05:
You could always try coding it :). The SHT3X Code is a good example.

closed 2018-07-04 by stale[bot]