arendst/Tasmota #2264: 5.12.0i strange issue

gthb96, 2018-03-26:
Make sure these boxes are checked before submitting your issue - Thank you!

Hello I am using a Sonoff Basic and just compiled version 5.12.0i via Platformio.
After I uploaded it via the web IF to the Sonoff Basic I was unable to access it via the webbrowser or it was very slow. It also was just randomly replying to pings.
After I downgraded to your build of 5.12.0 it worked perfectly again.

I was compiling 5.12.0i again and uploaded it via serial connection.
Now it was working fine, I was able to use the Web IF and also was able to ping it while it was still connected via serial adapter.
As soon as I connected it to the power outlet the same issue was back.

My question is, why is 5.12.0i just working on my Sonoff when I connect it via 3,3V directly but it works very slow and has no GUI when its on AC power?

5.12.0 was working on AC properly.

jcullen86, 2018-03-27:
I seem to have the same issue here, connects to wifi when connected to the FTDI adaptor but not to AC power. 
Still investigating at this stage.

Milbert-R, 2018-03-27:
I also have this problem. Only with Sonoff Basic. Sonoff S20 works fine.

jpadie, 2018-03-27:
dangerous, but have you done any serial sniffing whilst powered over AC?  i'd guess there are brownouts going on.

gthb96, 2018-03-27:
But why would it just be with the latest firmware version?

jpadie, 2018-03-27:
Perhaps because something in this version is drawing more power than previous?  Do a full diff of the repo and see whether you can see anything that might trigger this.

jcullen86, 2018-03-27:
Mine turned out to be a AC wiring issue. I connected a sonoff basic running 4.05 and it too failed to get an IP address. I cut about 3 inches off my wiring and reconnected and it's OK now ¯\_(ツ)_/¯

gthb96, 2018-03-28:
Does anyone else have this problem now too and its not just a wiring issue?

almostlunatic, 2018-03-30:
I can report this very issue I started to experiment yesterday with the usertimers fork of flexiti, based on the same build. The difference is that I do not use the AC, only 3.3v through the FTDI adapter. So, the usage of Sonoff Basic with this build is impossible. At first I thought the module was burned out but then I saw some sudden pings coming through. Downgrading to 5.12.0 changed the whole thing. I hope I could compare 2 codes tomorrow.

closed 2018-06-20 by stale[bot]