arendst/Tasmota #2266: Thingsboard

whhsn, 2018-03-27:
Hi,

I am looking for a local dashboard for Windows.

I am trying [thingsboard.io](thingsboard.io) but I can't get it to work with Tasmota.
Any suggestions for another dashboard or hints to read data from Tasmota.

Thanks,
WH

brianredbeard, 2018-03-28:
What is the concern?  thingsboard.io definitely has [a component][iot-gateway] which can process information from an MQTT broker.

[iot-gateway]: https://thingsboard.io/docs/iot-gateway/

Frogmore42, 2018-03-28:
What is your goal?  Thingsboard.io looks quite complicated and the documentation on mqtt did not look easy to follow.  It also looks cloud based, but I did not look too deeply.

Is your Windows goal for viewing only or are you planning on having a Windows system running 24x365?

whhsn, 2018-03-28:
Eventually we need to go to the cloud but for development, having the it on the PC is a plus!

The documentation on MQTT gateway is vague & too complected for me!

One problem is that you must publish to a certain topic:   "v1/devices/me/telemetry"
I can't figure out how to do that in Tasmota since the topic cannot be fully defined & there is the CMND, STAT, ... added to any topic.

Is there a way to

Frogmore42, 2018-03-28:
That doesn't really answer the question of what it is that you are trying to achieve.
While it is probably possible to get Tasmota to publish some stuff to a particular topic (the wiki covers this for switch topic as I recall) it probably won't do everything that you want.

NodeRed and/or one of the Home Automation tools are things that other people have used to do home automation type things.  Some of these can run on a PC (I run mosquito and Node-Red on a PC in my house) others have much better support on Linux and can run on a Raspberry Pi.

whhsn, 2018-03-29:
I am trying to find a good MQTT dashboard that can be local/cloud
Any suggestions are welcomed :)

Frogmore42, 2018-03-29:
Still doesn't answer the question.  What do you want the dashboard to show/control?

As an example, when I started my goal was to control the ventilation fan in my bathroom.  I wanted the fan to come on automatically when someone is taking a shower and turn off when the humidity had stabilized.  To do this I have a TH16 with a SHT-15 temp/humidity sensor talking MQTT, a local MQTT broker, a local instance of NodeRed, and emoncms.org for visualization.  I also have Alexa hooked up, so I can ask her to turn on the fan (though she has a hard time understanding "bathfan" she usually gets just "fan").

That is a pretty trivial goal, but it still took a decent amount of time to set it up.  It might have been easier if I had used a home automation solution, but they all looked complicated and somewhat limited, while I was pretty sure NodeRed could do just about anything and already had a lot of experience using emoncms.