arendst/Tasmota #2267: New PCB Layout SONOFF T1 R2 EU

pilzandreas, 2018-03-27:
New PCB Layout SONOFF T1 R2 EU:

You may update the wiki on https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU

Here is a foto of the PCB: https://photos.app.goo.gl/KHc7ls8YhdCAZQGF2

kellertobias, 2018-09-10:
Just bought the same device. Mine Looks the same. The description in the wiki for T1 or Touch doesn't work

pilzandreas, 2018-09-10:
@Tobias - eveything worked fine for me, which problems are You experiencing?

kellertobias, 2018-09-10:
works now also. all the way on the bottom of the page is a description. However, it does only work if I press the reset switch not too long.

AsWaM, 2019-04-02:
@pilzandreas   @kellertobias 

Hello, I got to of them, i can't get them in flash mode. How did you manage to do so?

Already flashed 5 with the old board, but this one, i have tried hundreds of time without success...

apilz, 2019-04-02:
[image]

the blue cable is Y connected to board and pin (right picture)

AsWaM, 2019-04-03:
Thank you so much! Thank you! It worked!!!