arendst/Tasmota #2268: Can't enter RF learning mode on T1; press+hold button 1 resets unit

jpadie, 2018-03-27:
using 2Ch T1 EU (R2) sonoff with tasmota 5.12.0.i

with sonoff firmware the device learns the remote control buttons just fine.  Flashed tasmota 5.12.0.i and I'm seeing a number of issues.  To start with the configuration via the web page was not being properly saved or was being ignored on startup.  hardcoding into settings.ino resolves this.

the rf learnings have been lost though.  and press and hold of button 1 to enter learning mode results in a beep then a reset.  here's the serial dump (some information redacted).  the longpress occurs at 15:16:36:

```
00:00:00 Project phenix T1Touch (Topic T1, Fallback phenixTouchController, GroupTopic sonoffs) Version 5.
12.0i-2_4_0
00:00:00 WIF: Connecting to AP1 *** in mode 11N as T1-3977...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on T1-3977.local with IP address 192.168.0.29
15:16:27 MQT: Attempting connection...
15:16:27 MQT: Connected
15:16:27 MQT: phenix/tel/T1/LWT = Online (retained)
15:16:27 MQT: phenix/get/T1/POWER =
15:16:27 MQT: phenix/tel/T1/INFO1 = {"Module":"Sonoff Basic","Version":"5.12.0i","FallbackTopic":"phenixT
ouchController","GroupTopic":"sonoffs"}

15:16:27 MQT: phenix/tel/T1/INFO2 = {"WebServerMode":"Admin","Hostname":"T1-3977","IPAddress":"192.168.0.29"}
15:16:27 MQT: phenix/tel/T1/INFO3 = {"RestartReason":"Software/System restart"}
15:16:27 MQT: homeassistant/light/T1_1/config =  (retained)
15:16:27 MQT: homeassistant/switch/T1_1/config =  (retained)
15:16:29 MQT: phenix/inf/T1/RESULT = {"POWER":"OFF"}
15:16:29 MQT: phenix/inf/T1/POWER = OFF
15:16:36 MQT: phenix/tel/T1/STATE = {"Time":"2018-0327T15:16:36","Uptime":"0T00:00:16","Vcc":3.469,"POWER":"OFF","Wifi":{"AP":1,"SSId":"***","RSSI":100,"APMac":"***"}}
15:17:04 MQT: phenix/inf/T1/RESULT = {"Reset":"Reset and Restarting"}
15:17:04 CFG: Use defaults
15:17:06 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16

tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld 
```
is it intended behaviour that tasmota resets and restarts on a long press of button 1?  If so how can RF learn mode be entered?  If not, any tips on where to look in the source code for a bug?

jpadie, 2018-03-27:
so, I see that there is an unhandled race condition here.  
from this page: https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

> Pressing the button for over four seconds: Reset settings to defaults as defined in user_config.h and restarts the device

and from this page https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU

> The Tasmota firmware is compatible with the RF remote feature built in to the T1 modules. Pairing a new remote is the same procedure as a stock Sonoff, I.E. hold the desired button on your sonoff until you are given 2 beeps (approx 7 seconds), press the button on your RF remote. The Sonoff will beep to confirm the pairing.

so the tasmota firmware leads to a race condition whereby you can never pair the primary button.  

I will look for a fix unless anyone has an explanation for this design and a known workaround.  

thanks

arendst, 2018-03-27:
If I were you I would take a look in the wiki regarding available commands.

But then, I'm not..

jpadie, 2018-03-27:
nope. only i’m me.

i did take a look.  I’m not talking about the serial or web console.  there may be other routes - the issue i reported is the race condition on the button press for learning RF.  I believe that the issue has its route around line 1860 of sonoff.ino.

emontnemery, 2018-03-28:
You need to send 'setoption13 1', this disables multi-click functionality and makes timeout for long press much longer.
Please close this issue.

jpadie, 2018-03-28:
that's not a solution;  it's an unattractive work-around.  
and in my case has not worked: the device still does not enter learning mode.  

the issue should remain open, in my opinion.

emontnemery, 2018-03-28:
Why is it an "unattractive workaround"? Long press functionality in Tasmota firmware is fundamentally incompatible with the HW-controlled RF learning mode in the Sonoff T1.
Are you sure you sent command "setoption13 1" and it was stored?
You are not running the T1 without the relay board connected?

jpadie, 2018-03-28:
It is unattractive because the concept of ‘hold’ is a good one. 

But key here is that either we must acknowledge that there is a bug, or that the wiki is wrong/misleading/incomplete in its statement as to how RF learning mode is entered.  

on the device I’m testing the relay board is attached.  

the button options are currently as follows:

{"SetOption11":"OFF"}
{"SetOption12":"OFF"}
{"SetOption13":"ON"}
{"SetOption14":"OFF"}
{"SetOption15":"ON"}
{"SetOption32":"100”}
{"SwitchMode1":0}
{"SwitchMode2”:1

a long press (until the double beep) no longer resets the board.  So that’s a positive.  

During a long (7 second) press until the double beep, the serial console shows that pubs are being sent for the press event.  I suppose this is intended behaviour.

However the device does _not_ enter RF learning mode after the double bleep.  I am testing with five different remotes all successfully paired to the same device before flashing with Tasmota.

emontnemery, 2018-03-28:
There is **no way** for Tasmota firmware to alter the behavior of the RF learning, as that's entirely handled by an external micro controller.

Why do you set option15 to 1, that's for using PWM as LED dimmer. Sonoff T1 has no PWM function, only relay.

Also, please read the manual, double beep means "unlearn", not "learn".

jpadie, 2018-03-28:
i have not explicitly set option15.  that is what came up when querying the interface.  It is unexpected.

Please refer to the wiki - it explicitly states that the RF learn mode requires seven seconds and a double beep.  If the wiki is wrong, then that further underlines my point.

As for your assertion - I understand what you are saying.  I am surprised as well.  I have just flashed another firmware (espurna) and that exhibited the same symptoms.  Plugged into the mains (and disconnected from serial of course), the learn function appears to work - which suggests that the radio needs more than 3.6v to work.  

However it does not toggle the switch or fire off a mqtt_pub. so is useless.

I am re-flashing back to tasmota to see what the behaviour is there.

emontnemery, 2018-03-28:
Maybe I was unclear earlier: The Sonoff T1 does not work if not connected to mains and without relay board connected. I'll update the wiki to make it more clear, many users are hitting the same issues as you do.

jpadie, 2018-03-28:
the relay board was connected.  I was powering it with a high-power supply to avoid brown outs.

emontnemery, 2018-03-28:
Yeah, but the relay board supplies higher voltage than 3.3V to the main board, so just supplying power through the debug conenctor won't do it.
Anyway, can you confirm your problem is solved when connecting to mains?

jpadie, 2018-03-28:
not yet.  

Just having supper then will finish flashing tasmota and check that it works.

jpadie, 2018-03-29:
Am giving up for the day.  Just getting garbage out of the device serial console after multiple reflashes trying to fix.

`in the restartin the restartin the restartin the restartin the restartin the restartin the restartin the
restartin the restartin the restartin the restartin the restartin the restartin the restartin the restart
in the restartin the restart
`

I will clean down the environment again, rebuild and reflash using a different machine.  I will post back tomorrow evening when i next have time to work on this.

jpadie, 2018-03-29:
that's fixed now.  the build is clean and the RF pairing now works.  

I don't think the ability to reset via a long press of only a few seconds is sensible on a Touch device like this.  I'd suggest that the build flags push that out to (say) 45 seconds for this class of device.  

thanks.

arendst, 2018-03-29:
Consider use of SetOption1 as soon as the device has been configured to your expectations. This options disables the restart on hold and more than 2 multi presses to select initial wifimanager.

jpadie, 2018-03-29:
thanks.  

i think i will need to set these in the firmware rather than via the interface.

emontnemery, 2018-03-29:
@arendst There have been so many questions/issues about Sonoff T1 resetting due to unintentional long press..
The debug prints when resetting due to hold press could be a bit more clear, something like:
```
if (!Settings.flag.button_restrict) {   // No button restriction
    snprintf_P(scmnd, sizeof(scmnd), PSTR(D_CMND_RESET " 1"));
    AddLog_P(LOG_LEVEL_INFO, "Long press detected, reloading defaults settings");
    ExecuteCommand(scmnd);
} else {
    send_button_power(0, button_index +1, 3);        // Execute Hold command via MQTT if ButtonTopic is set
}
```
Also, maybe module template should default to `setoption1` set to 1 for wall switch type modules.

Should I submit a PR?

arendst, 2018-03-29:
It can't be set to default 1 as initially users still have to be able to select wifi manager option as long as they do not understand to the use of the serial interface (as most do considering the amount of butten press issues they have).

I was considering extending the hold time for the specific reset option but then the Dual R1 will fail as it has a fixed amount of time (4 seconds) before it's code changes from once pressed to hold.

Being a one size fits all is just being a bitch to some.

jpadie, 2018-03-29:
Perhaps automatically defaulting to WiFi manager if no ap is found?

sverspecht, 2018-04-12:
> There have been so many questions/issues about Sonoff T1 resetting due to unintentional long press..

Yes, I've made bad experience after successfully installed Tasmota "Over The Air" on my Sonoff Pow, linked to my MQTT broker and controlled it via Homebridge. Now after, having hold too much time button, my Sonoff seems to be completely freezed. He doesn't connect to Wifi and doesn't awswer when pressing button (neither short or long press.

Any idea to fix this without FTDI?

ascillato, 2018-06-02:
Hi @jpadie 

If your issue is solved, please close it. Thanks! :+1:

closed 2018-06-27 by ascillato2