arendst/Tasmota #2271: Impossible Specify single led color in Clock Mode

Giorgiofox, 2018-03-28:
Actually is not possible set a single led color in clock mode, but I think this can be a very useful function for show information trough MQTT, such as timers, informations, notifications, etc.

Can be implemented?

arendst, 2018-03-29:
If you do not select clock mode you can address every led individually using the led command.

Giorgiofox, 2018-03-29:
Yes I know! But I'm kindly asking to implement this also in clock mode ;)

nightcat91, 2019-06-24:
I would like to get this looked at again.
It is still correct that you can use the commands to command a single led to display a certain color, but this cannot be done while in clock mode.
I would like to be able to do that to make small markers like on a real clock.

Hope someone can implement this.

The result could look something like this:
https://youtu.be/wYYCyfEK_Rw?t=83